micropython/micropython #1036: Stackless design?

pfalcon, 2015-01-03:
@dpgeorge, did you consider stackless design? I.e. one where Python native function call wouldn't map to C function call, but rather there was big continuous value stack?

pfalcon, 2015-01-03:
Why asking: C stack design essentially prevents normal tracing JIT implementation. That's because normal tracing JIT recurses into called functions and inlines them, but then if JIT trace aborts inside such recursed function, JIT has to reconstruct stack frames leading into that function. That's boring on its own, but would great chore and very hard to port if done on C stack.

dpgeorge, 2015-01-03:
The word "stackless" confuses me, because you always need some kind of stack, on both the Python and C sides (the latter, at least for pushing RIP and RSP).

I wanted to use the C stack as much as possible so that you don't need dynamic memory allocation for basic calls.  Yes, you could allocate a separate Python-value stack at the beginning of the program, but why have 2 separate stacks (which are never going to be optimal in size and will waste RAM) when you can get away with 1?

A Python-value stack has its advantages; eg you then have better knowledge of root pointers for the GC scan.  But I think it's faster to just pass Python values as arguments to functions, rather than push/pop all the time.

So, my design decision was based on minimal dynamic RAM usage, and speed.

What exactly do you mean by "stackless design", in this context?  Do you mean that a bytecode function calling another bytecode function uses the same VM call (instead of going through a few C functions and making another VM stack frame)?

pfalcon, 2015-01-03:
> The word "stackless" confuses me

Yeah, it's used in the same sense as http://www.stackless.com/ , and they mean "not using C stack" by it.

pfalcon, 2015-01-03:
> Do you mean that a bytecode function calling another bytecode function uses the same VM call (instead of going through a few C functions and making another VM stack frame)?

Yes, exactly that. So, CALL_FUNC would create another frame on Python stack, switch to it, and keep running bytecode, while RETURN would pop to previous frame.

pfalcon, 2015-01-03:
I wanted to know if you had specific anto-reasons to implement it like that. Yes, it means more detailed memory management for stack frames. The benefit is that it makes VM more self-contained and more reflective.

And thanks for the answer - holidays are running out, and I'd really like to have some outcome in my head on VM design research I did these couple of weeks ;-).

dpgeorge, 2015-01-03:
> Yes, exactly that. So, CALL_FUNC would create another frame on Python stack, switch to it, and keep running bytecode, while RETURN would pop to previous frame.

Ok.  Implementing such a feature would be complicated by the fact that CALL_FUNC would need to check first if it's a bytecode call or a native call.

> I wanted to know if you had specific anto-reasons to implement it like that.

Not really.  Previous VMs I implemented (for other languages) did it this way, but then they had special opcodes for native/C calls.  It's just simpler to have it the way it is now, and it's seamless to call C from Python and Python from C (since it's just mp_call_function that does both).

> Yes, it means more detailed memory management for stack frames.

Yep, would need a larger C stack and more liberal use of alloca, or a separate (and large) Python value stack.

> The benefit is that it makes VM more self-contained and more reflective.

And also faster I would guess.  And it'd use less RAM for deeply nested Python-Python calls.

dpgeorge, 2015-01-03:
> I'd really like to have some outcome in my head on VM design research I did these couple of weeks ;-).

Did you see Wren: https://github.com/munificent/wren and http://munificent.github.io/wren/performance.html

There are many directions we can go with the VM.  Remember though that `@micropython.native` already buys you around 2x speed up :)  What is the goal here, speed, RAM usage, ...?

pfalcon, 2015-01-03:
> would be complicated by the fact that CALL_FUNC would need to check first if it's a bytecode call or a native call.

Surely, that's not too big a complication to peak at function types ;-).

---

So, considering that C stack usage an optimization, it's ironical that to allow more advanced optimizations on native side (like tracing JIT), it takes to de-optimize interpreter by clearly separating C and interpreter control state. It's tempting then to go for more optimizations - for example, dedicated value stack enables precise garbage collection - after all, it will be the root set then, and all values there are tagged. But that's exactly will be detrimental for native code.

pfalcon, 2015-01-03:
> What is the goal here, speed, RAM usage, ...?

I'm trying to dispel LuaJIT magic for myself. @micropython.native's speedup of 2x is too small. They say LuaJIT is 200% faster than C (well, something like that :-D). There's whole subculture around LuaJIT, fueled by people trying to make it something beyond bent Lua semantics, not being able to, and the guy keeping repeating that all the magic is enabled by Lua. So, I'm trying to understand how tracing JIT works, play with some code, and confirm that this capability can be added to uPy. The end applied goal is performance thus, but compiler optimizations is actually my long-term theoretical interest.

pfalcon, 2015-01-03:
Thanks for link to Wren, will look into it.

dpgeorge, 2015-01-03:
Insipred by Wren performance page I ran method_call.py on uPy and was disappointed; we only do as good as CPython, even with native mode.  So I ran gprof and the culprit is the obvious one: hashing and map lookup.
```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  us/call  us/call  name    
 25.68      0.38     0.38 23666922     0.02     0.02  mp_map_lookup
 24.32      0.74     0.36  4333341     0.08     0.27  mp_execute_bytecode
  9.46      0.88     0.14  4333341     0.03     0.03  mp_setup_code_state
  4.39      0.95     0.07 23333392     0.00     0.00  qstr_hash
  4.39      1.01     0.07 23333392     0.00     0.01  mp_obj_hash
  3.38      1.06     0.05  7000008     0.01     0.04  mp_obj_class_lookup
  3.38      1.11     0.05  2400005     0.02     0.02  mp_binary_op
  2.70      1.15     0.04  4333341     0.01     0.05  fun_bc_call
  2.36      1.19     0.04  4333338     0.01     0.01  mp_call_method_n_kw
  2.03      1.22     0.03 10333333     0.00     0.04  instance_load_attr
  2.03      1.25     0.03                             nlr_push
  1.69      1.27     0.03  6333333     0.00     0.04  mp_load_attr
  1.69      1.30     0.03  4333336     0.01     0.01  mp_obj_is_callable
  1.69      1.32     0.03                             printf_wrapper
  1.35      1.34     0.02  4333336     0.00     0.05  mp_load_method
  1.35      1.36     0.02  2666670     0.01     0.07  instance_store_attr
  1.35      1.38     0.02  2666670     0.01     0.08  mp_store_attr
```
Definitely something worth looking further into.

pfalcon, 2015-01-03:
Had a look at Wren. Nothing magical. Well, docs are good, makes it easy to assess it ;-). Float NaN tagging is old trick, LuaJIT uses it too. It caters for dishonest CPUs with too many bits and crunching floats in one cycle as ints. But you can't build such CPUs on kitchen table. And all the rest CPUs will suffer 2x memory usage. So, that scheme clearly not suitable (as the only one) for us. Of course, it would be cute to have it as a configuration option. I'm afraid, that will realistically require C++-level abstraction to be manageable.

The rest of things like more static typing is what I'd like to see in uPy (with explicit enable option).

Ah, and on topic of this ticket, it uses value stack for function calls. Stack implemented as single array (each frame not allocated on heap). Squirrel used this design too. I didn't look into Lua implementation, based on docs, data model for it is individual frames, but dunno if they're allocated on heap or in static array.

Coroutine support and reduced memory fragmentation of course cater for "stack of stack frames" design, vs big static array which is pain to reallocate.

pfalcon, 2015-01-03:
> the culprit is the obvious one: hashing and map lookup

Yeah, maps are really slow in uPy, other side of extreme memory efficiency. We should add support for non-1.0 load factor, for "fast" builds, that should alleviate it a lot. Caching hashes in maps may help too (but I still not sure I like bigger-than-a-byte hashes, again, would be nice to have that configurable ;-) ).

dpgeorge, 2015-01-03:
> (but I still not sure I like bigger-than-a-byte hashes, again, would be nice to have that configurable ;-) )

Off-topic: I have a patch in the queue that removes explicit qstr hashes completely and just uses 1 least-sig byte of the length, and first byte of the string, as the hash.  This saves around 500 bytes ROM (and will save RAM), and can be used for space-constrained builds.  Did not yet assess speed implications of this patch.

dpgeorge, 2015-01-03:
For the record, NaN tagging without bignum leads to silliness such as:
```lua
lua > print(0xffffffffffffff-0xfffffffffffff0)
16
```
when it should be:
```python
uPy >>> print(0xffffffffffffff-0xfffffffffffff0)
15
```

pfalcon, 2015-01-03:
> I have a patch in the queue that removes explicit qstr hashes 

Configurable I hope? length + first byte is pretty bad hash ;-).

dpgeorge, 2015-01-03:
> Configurable I hope?

It will be.  That's why it's still in the queue!

pfalcon, 2015-01-03:
> For the record, NaN tagging without bignum leads to silliness such as:

Yes, sure, using tagged double would mean having much reduced int range on 64-bit platform (than by 1 bit as in uPy). That would need to be compensated by bigints.

I'd say causation in your example is the opposite: it works like that because core Lua semantics lacks integer type at all. They were able to hack it in using tagged NaNs, but it has limited utility - AFAIK, for Lua mainline it's used only to implement bitwise ops, and LuaJIT uses it only for platforms w/o hardware FP.

pfalcon, 2015-01-03:
Ok, thanks for the discussion, adding this "stackless" design to things I'd like to try for uPy ;-).

Closing.

closed 2015-01-03 by pfalcon

pfalcon, 2015-01-04:
Btw, http://wingolog.org/archives/2011/05/18/value-representation-in-javascript-implementations gives another issue with NaN-tagging on 32-bit systems and conservative garbage collection: increased probability of pointer mis-detection, as both parts of otherwise tagged value will be scanned separately.

dpgeorge, 2015-01-04:
Interesting article.  Relevant also for us is the comment there that on 32-bit systems passing around 64-bit values for each argument and return value can be costly.

pfalcon, 2015-02-15:
@dpgeorge, FYI, I now have proof of concept implementation of this "stackless" feature, where bytecode function call frame is allocated on heap instead of C stack (and thus calling any func requires mem alloc). That was relatively straightforward. The biggest issue is of course plethora of calling methods - I so far tested it only with MP_BC_CALL_FUNCTION.

dpgeorge, 2015-02-15:
Is there a way to not need heap allocation?  Switching to stackless in this way will completely wipe out all the hard work making uPy run without needing heap allocation for basic things...

pfalcon, 2015-02-16:
Well, just as Stackless Python, I don't intend these "stackless" patches to replace normal uPy schedule, just be a separate feature with own usecases. But I hope they will be mergeable upstream (and selectable by a config option), as it all goes pretty neat, and patches are not too obtrusive (well, of course, it requires splitting existing (complex) functions, which may lead to extra stack usage to pass args among them, but that should be compiler's job to inline them back ;-) ).

dpgeorge, 2015-02-16:
I'm all for configurability.  We will have "stackless" vs "heapless" :)