micropython/micropython #1037: windows version VS 2013 community edition

denfromufa, 2015-01-04:
i get the following errors after 'git init' in VS project folder:

// This file was generated by genhdr.targets
#define MICROPY_GIT_TAG "fatal: bad revision 'HEAD'"
#define MICROPY_GIT_HASH "unknown-dirty"
#define MICROPY_BUILD_DATE "2015-01-04"

// This file was generated by genhdr.targets
#define MICROPY_GIT_TAG "fatal: bad revision 'HEAD'"
#define MICROPY_GIT_HASH "unknown-dirty"
#define MICROPY_BUILD_DATE "2015-01-04"

Error	1	error LNK2019: unresolved external symbol ___builtin_expect referenced in function _mp_obj_str_format	C:\Python\micropython\windows\objstr.obj	micropython
Error	2	error LNK1120: 1 unresolved externals	C:\Python\micropython\windows\micropython.exe	1	1	micropython

stinos, 2015-01-04:
Add these lines to windows/mpconfigport.h:
```
#define MP_LIKELY(x)                (x)
#define MP_UNLIKELY(x)              (x)
```

I'll create a PR for it.

pfalcon, 2015-01-04:
Should be fixed via https://github.com/micropython/micropython/pull/1039

closed 2015-01-04 by pfalcon