micropython/micropython #1040: TypeError in bytearray initialization

kfricke, 2015-01-04:
As [suggested on the forum](http://forum.micropython.org/viewtopic.php?f=3&t=471#p2675), here is a bug report for the new bytearray implementation. When initialized all versions of uPy starting from 1.3.8 do throw a TypeError:
  `TypeError: 'int' object is not iterable`

@pfalcon suggested to use the following test case to describe and reproduce this bug:
  `bytearray(2**65 - (2**65 - 1))`

pfalcon, 2015-01-04:
Thanks, the issue as exposed by the snippet above is fixed in https://github.com/micropython/micropython/commit/343ca1e63ae90b13af8d0a71f6ed6fc5a6284e33 . The caveat is that I'm not sure how you could get a long int there in your original script, so please test if this really fixes your issue.

kfricke, 2015-01-04:
The value I do initialize the bytearray with is 513 in my script

pfalcon, 2015-01-04:
Yes, it shouldn't lead to the error you report (and that part of code didn't change lately). And yet that's the only reason (as addressed by the patch) I can think of why that issue may occur.

kfricke, 2015-01-04:
One variable of the arithmetic is the result of an int() conversion (`self.pages = int(self.height / 8)`). This is absolutely useless imho, but taken from the original [SSD1306 library](https://github.com/khenderick/micropython-drivers).

pfalcon, 2015-01-04:
Ah, but of course. `self.height / 8` is a float, and we conservatively convert float to long int. And the real fix is to use integer division: `self.height // 8`.

So, mystery solved, I'm sure this fixes it then, closing. Please reopen if anything.

closed 2015-01-04 by pfalcon

kfricke, 2015-01-04:
Thanks a lot for this explanation!

kfricke, 2015-01-04:
Cool it even works ;-)

dpgeorge, 2015-01-04:
> Ah, but of course. self.height / 8 is a float, and we conservatively convert float to long int. And the real fix is to use integer division: self.height // 8.

Nicely solved.

But it actually points to some inefficiencies introduced by 12033df511a5a5f0237a764379aacfbceb6c45dd.  I use quite often something like `int(x * a / b)` which used to give just a small int, but now always returns a long int, even when most likely the number will fit into a small int.  We should probably add a small-int path to either mp_obj_int_make_new or mp_obj_new_int_from_float to return a small int if the float value fits.

pfalcon, 2015-01-04:
> some inefficiencies introduced by 12033df

I can share the sentiment - I used only small int check/accessor in bytearray() because I though that realistically it always will be such. But now that I lay high efficiency hopes on tracing JIT, robustness in the interpreter looks more important ;-).

Added comment to https://github.com/micropython/micropython/pull/1033#issuecomment-68641583

doogle, 2015-01-05:
Hi! @pfalcon directed me here. As @dpgeorge has said, it looks like a test should be added to the various `mp_obj_new_int_from_float()s` to use `MP_OBJ_NEW_SMALL_INT` where possible.

Of course one approach would be to use frexp() here, but as it's been decided that avoiding this (and going with a bitbashed approach) is ok inside `mpz_set_from_float()`, then this test might as well use the float bits directly as well. Does that make sense?

Regarding each variation of `mp_obj_new_int_from_float()`... in both objint.c and objint_longlong.c there is still the TODO to raise an exception if the int won't fit. And in objint_longlong.c and objint_mpz.c, there is the requirement to create a small int if possible.  
Therefore I think a function is needed that can return the number of bits needed in order to represent a given mp_float_t as an integer (sort of a `round(log2())`, but more optimal for this case). The caller can then use this value to decide on its behaviour.

My proposal would be to:

1. Rename and move EXP_SZ/FRC_SZ/mp_float_int_t to mpconfig.h (as they'll be used in multiple places). The union could even be defined in a header (if so, where would make sense?). Also, I've realised that EXP_SZ/FRC_SZ should probably use float.h's *_MAX_EXP and *_MANT_DIG rather than hardcoded values (sorry).

2. Add the previously described function, maybe in objint.c so it can be used by all objint*.c.

3. Use the function in each objint*.c's `mp_obj_new_int_from_float()` to decide whether to raise an exception, convert to a small int or convert to a long/mpz int.

How does this sound?