micropython/micropython #1041: TODO for native codegeneration (@micropython.native)

pfalcon, 2015-01-05:
I would like to capture my findings from looking into native codegeneration. Things yet to do in rough order of priority:

- [ ] Support functions with more than 3 args
- [ ] Support functions with *args
- [ ] Support functions with **kwargs
- [ ] Support "with" statement
- [ ] Support raise w/o args
- [ ] Support generators

First three, related to function argument processing are the most important of course. Supporting more than 3 (or N) arguments brings the question how to implement them:

1. Follow native C calling conventions?
 - That requires still putting high limit on the number of supported args, or
 - implement something like FFI to avoid implementing it as a switch casting it to function pointers with appropriate # of args
2. Follow the same approach as for builtin functions - after some # of args, pass them as an array, by pointer.

dpgeorge, 2015-01-05:
Generators are difficult because you need to make a proper closure (ie a C stack on the heap with RIP stored).

The first 3 points I think can be done in a relatively straight forward way, reusing lots of existing code: if more than 3 args, or kwargs are given, then use mp_setup_code_state to do the processing and create the args in some state, then just pass this state directly to the native Py function (so implementing your point 2 above instead of 1).  We would also probably keep the current code (0, 1, 2 or 3 args) for a fast path.

pfalcon, 2015-01-15:
> Generators are difficult because you need to make a proper closure

Yes.

> (ie a C stack on the heap with RIP stored).

No, because:

> create the args in some state, then just pass this state directly to the native Py function

Generally, frame is passed to a function. We just got used that it's passed implicitly, via SP. But for languages with closures, it needs to be passed explicitly (can still be allocated on stack as an optimization). It would be fun to just pass it explicitly in BP, but that breaks C calling convention. So, just have to pass frame pointer as 1st C arg, and all native args are passed via frame. That's base form of implementing a call. Then yes, calling a non-closure with limited number of args can be optimized to follow C calling conv for args directly. So far, we have only this optimization in place (closures (not just generators) don't work with native code, right?)

Otherwise, people appear to keep assuming that arbitrary C function can yield. Surely, it can't. And I don't see a need to yield from raw C code at all. But even if there will be such need, such C function will be implemented per the schedule above - it will take explicit frame pointer, and that's where it keeps all its locals. And when it calls another function which may yield, it allocates frame for it before call. And in the programming model we discuss (and that's the programming model of Python, C#, C++, but not of Lua, Stackless Python, Python Greenlets) it's always known which function may yield, and user is required to call it in a special syntactic manner (mark call with "yield from" in Python, "await" in C#/C++). The end result is that it's always known which function yields, it can be accounted, optimized, and stack munging is not required in this model. (But delegating to sub-coroutines requires special provision in this model - at least from slides in https://github.com/micropython/micropython/pull/1058#issuecomment-70047530 it's unclear how C++ is going to handle this issue).

dpgeorge, 2015-01-15:
> closures (not just generators) don't work with native code, right?

Right.  But closures are almost trivial to implement, compared with other features above.

> But even if there will be such need, such C function will be implemented per the schedule above - it will take explicit frame pointer, and that's where it keeps all its locals.

Native Python function that yields must also save its instruction pointer, or else have some code at the start of the function to jump to the correct location when its restarted.

pfalcon, 2015-01-15:
> Native Python function that yields must also save its instruction pointer, or else have some code at the start of the function to jump to the correct location when its restarted.

Sure, just like bytecode functions have `self->code_state.ip`, so not much difference here.

dpgeorge, 2015-04-07:
In 9988618e0e0f5c319e31b135d993e22efb593093 I implemented full support for function arguments (keyword, default, etc) in the native emitter, for x86, x64 and thumb2 (not ARM...).  It's not as neat as I'd hoped it would be but works and is relatively efficient in stack usage.  It reuses a lot of the existing bytecode functionality to do the hard work of setting up the args.  It can be optimised and/or cleaned up in the future if needed.

So... now I could run pystone with the native emitter... and the results are disappointing: the native emitter is slower than bytecode!  This is because the bytecode implements cached dict lookups for attr's and globals, whereas the native emitter does not, and this is an important factor in the pystone benchmark (and likely a lot of other benchmarks).  It will be a bit of work to get cached dict lookups working with the native emitter.  This is certainly not a priority :)

FWIW: without caching bytecode gets 60k pystones and native gets 75k, so there is some potential speedup.

pfalcon, 2015-04-08:
@dpgeorge , Great work! Thanks for doing that, I looked at native codegen, but as I mentioned, didn't get much of its internal workings, so switched to other directions.

The speed up may vary, that's understandable. But this change is important step towards being able to stick decorator a random function and being able to measure how much it helps, without need to additionally "porting" it.

dpgeorge, 2018-09-04:
Support for unwinding jumps in native code (return/break/continue within try/with) was added in 4ae7111573f6866e027e2919e4d331c01108db23

dpgeorge, 2018-10-01:
Support for generators in native code (yield and yield-from, and some parts of async) was added in cc2bd63c57a72ec37d839965c6bb61cd1fa202fc

pfalcon, 2018-10-31:
Continuing from https://github.com/micropython/micropython/issues/4268#issuecomment-434138251 

> If there are concrete proposals for how to do dict lookup caching in native code that would be interesting, but still not a priority to work on.

So, it makes starting with say that self-modifying code is of course a big PITA on modern systems due to instruction barriers, etc. The ideas are:

1. Whole-program analysis, to try to encode "right" indexes into the machine code at compile time, never change them. But whole-program analysis is separate big topic, far from being approached.
2. Then the idea is to go for self-modifying code after all, but do that at most once (I don't remember if bytecode caching may update value again and again, I'd assume yes). I.e. there would be `dict_get(obj, key, expected_idx)` call. Ideally, expected_idx should be set right before the call, and default value for it is 0.  If dict_get detects a miss, and expected_idx is 0, then it takes return address, and modifies instruction to set expected_idx for the next time.

dpgeorge, 2020-06-09:
The only remaining item here is to support raise without args in native code.  That's really difficult and it could be that it never gets implemented.  It's anyway rather easy to workaround, just use "except Exception as er" and then use "raise er".

The lack of support for raise without args is encoded in the test suite, so I'll close this issue for now.

closed 2020-06-09 by dpgeorge