micropython/micropython #1042: Optimise conversion of float to int

dpgeorge, 2015-01-05:
Currently, `int(3/2)` will create a long int, which is suboptimal.  See issue #1040 for background, especially https://github.com/micropython/micropython/issues/1040#issuecomment-68793436

@doogle so your proposed function will be like fpclassify, but it'll also return a value indicating if the float is small enough to fit in an int, right?

Remember that we need to think "micro", and so it's not always best to do the "right thing".  For example, a quick hack that will go a long way optimising this cas is to simply check if the floating point number is between, say, -1000.0 and 1000.0, and, if so, convert it to a small int.  Otherwise use a long int.  Note that due to floating point rounding errors we can't check that the float is between `MP_SMALL_INT_MIN` and `MP_SMALL_INT_MAX` (defined in py/smallint.h) (I tried it and it fails on the edge cases.)

I know some hardware saturates the float->int conversion on overflow, but that's not very portable.

How big is the function to check if a float will fit in an int?  Could we replace the standard fpclassify function with your new version, to eliminate duplicated code?

doogle, 2015-01-06:
In https://github.com/micropython/micropython/issues/1040#issuecomment-68793436 I described a function that would calculate the number of bits required to represent a given mp_float_t as an integer.  
I think you're right though... a better approach would be to have a replacement to `fpclassify()` that calculated which of these possible categories an mp_float_t fitted into:

1. Inf (would lead to an OverflowError exception)
2. NaN (would lead to a ValueError exception)
3. Fits into small int
4. Fits into long int (`MICROPY_LONGINT_IMPL` != `MICROPY_LONGINT_IMPL_NONE` only)
5. Doesn't fit (would lead to an OverflowError exception)

I've got some ideas for the classification function... I'll see what I can come up with in order to answer your last question. Obviously if it can't be done in an optimal way, a simple range check might just be the best solution.

I guess one other approach would be to use fenv.h's exception code functions, to decide on the validity of the float-to-int cast after the fact. What're your thoughts on that? AFAIK this might result in slightly different behaviour on different platforms.

pfalcon, 2015-01-07:
Well, we know that (IEEE) mantissa is normalized ad in range [1, 2). Then we can just check an exponent to see if resulting number can fit in the number of small_int bits. This check should be abstracted to a function (macro) and used once in mp_obj_int_make_new().

doogle, 2015-01-07:
I've submitted a pull request #1047 for this. Let me know what you think.

The code size for `mp_classify_fp_as_int()` is around 100 bytes for either x86 32 bit or 64 bit, with MICROPY_FLOAT_IMPL == MICROPY_FLOAT_IMPL_DOUBLE.

You might not agree with the changes I've made regarding the use of float.h and ieee754.h, in which I can go back to not using those headers.

@pfalcon to be accurate (which may not be necessary), there needs to be some extra logic to take into account negative numbers (as float->int cast rounds towards 0, but the mantissa is always positive), denormal numbers (<1), as well as inf and NaN.
The code could be made smaller by not caring so much about a few of the edge cases and just erring on the side of caution. I can make changes to this effect if desired.

doogle, 2015-01-07:
It seems most compilers don't have a ieee754.h. I'll revert to not needing this.

dhylands, 2015-01-07:
I noticed that as well when doing the floating point formatting code. So there are some constants near the top of formatfloat.c that should perhaps be moved into a common header?

doogle, 2015-01-07:
@dhylands it looks like those constants only cover single precision floats (ie. not doubles). Also, I've been using a union to access float bits, rather than explicit mask-and-shifts.

float.h provides enough information that I can create the union/struct and exponent bias constant myself, without needing ieee754.h. It's previously been stated that supporting mixed-endian doubles isn't a requirement for now, which simplifies things a bit as well.

The question is then is where would be the best place to define this union/struct? A new header, py/misc.h, or somewhere else? (since I'm still new to the codebase).

dpgeorge, 2015-01-24:
Now optimised; see 0fb17f6ef46ade018a3fda8d4266f15638ddf5e3

closed 2015-01-24 by dpgeorge