micropython/micropython #1044: Error from pyb.delay(int(n/2)) where n is in 0 or 1

MartyMacGyver, 2015-01-07:
This code:

```
import pyb
pyb.delay(int(1/2))     # pyb.delay(int(0/2)) has the same effect
```
leads to the following error:
```
OverflowError: overflow converting long int to machine word
```

pyb.delay(0) works fine though... 
I wouldn't expect it to fail and in fact it didn't until sometime during the last couple of weeks of builds.

*(Tested on Micro Python v1.3.8-49-gd8bfd77 on 2015-01-07; PYBv1.0 with STM32F405RG)*

MartyMacGyver, 2015-01-07:
Note: this error was first seen in build pybv10-2015-01-03-v1.3.8-41-g6fd4b36 (it is not present in pybv10-2015-01-02-v1.3.8-36-gfd40a9c)

dhylands, 2015-01-07:
I think that this is related to https://github.com/micropython/micropython/issues/1040#issuecomment-68641249

If you were to use integer division 1 // 2 or 0 // 2 then you would side-step the issue.

MartyMacGyver, 2015-01-07:
In my opinion it's a failed regression (and yet #1040 is closed). There are various ways to work around it, but I'm not really sure why those particular values (and not, say, 3/2) are triggering this and thus it's hard to say when the bug will crop up again. It's not the sort of error (or really side-effect) one would expect to see happening during basic division and integer conversion.

dpgeorge, 2015-01-07:
Yes, this is a bad regression.  Yes it's due to the "fix" introduced by #1040.  Yes it's strange that 1/2 breaks it but not 3/2.

dpgeorge, 2015-01-07:
@doogle I think this is a bug in mpz_set_from_float, when converting a number smaller than 1.  The following crashes on unix build (x64 at least):
```python
import struct
struct.pack('I',int(1/2))
```

doogle, 2015-01-07:
It looks like this is caused by my use of `mpz_init_zero()`. This results in `z->dig == NULL`, which then causes a segfault in `mpz_hash()`.

It seems `mpz_init_zero()` is the wrong function to use to set the number to 0. I'll submit a patch fixing this.

dpgeorge, 2015-01-07:
Fixed, and test cases added to catch this bug.  Fix is robust, but temporary, pending thoughts in #1042.

@MartyMacGyver Thanks for the report!

dpgeorge, 2015-01-07:
> It seems mpz_init_zero() is the wrong function to use to set the number to 0.

No, it's the right function, it's just broken!  (Or, the way functions assume things about z->dig is broken.)  In fact, it seems that so far an mpz "0" has never been returned from the mpz module, so this bug was never seen before.  Well, it's good that "0" has not needed to be returned, because we would expect only big ints to come out of mpz.

The mpz zero stuff needs thought and a proper fix.

dpgeorge, 2015-01-07:
@doogle probably can just replace `--d` with `d--` in those loops.

doogle, 2015-01-07:
I was just looking at how I would create a 'safe' mpz 0 and realised that it shouldn't require z->dig to be non-NULL.
I'll leave `mpz_set_from_float()` using `mpz_init_zero()` (I'm making other changes to it as part of #1042 though).

In `mpz_hash()`, although `d` should be NULL in the case of an mpz 0 (with both z->dig == NULL and z->len == 0), would it maybe just be safer to surround the `while` with a check on `z->dig != NULL`?
Although if that function is for calculating a hash, it looks like it needs some attention anyway.

MartyMacGyver, 2015-01-08:
Commit ad2307c92c15f0aa90dbd0741fd2538719d0b5e1 seems to have resolved this - I'm using pybv10-2015-01-08-v1.3.8-56-g181bfb6 now and I'm not seeing the issue.

closed 2015-01-10 by MartyMacGyver