micropython/micropython #1045: Erratic testsuite results running on float-less target

pfalcon, 2015-01-07:
Like ESP8266. Should we make sure that floats are not used in basic/*, and split all float-related testcases to float/ ?

dpgeorge, 2015-01-07:
> Should we make sure that floats are not used in basic/*, and split all float-related testcases to float/ ?

That's how it is supposed to be.

dpgeorge, 2015-01-08:
In 19b3fea6a885d071a1216614ee3f766aba947cbf I removed test cases from tests/basics/ that relied on float and put them in tests/float/.

All other tests that are not in tests/float/ dir do not rely on float, except the following:
- extmod/ujson_dumps.py
- extmod/ujson_loads.py
- misc/features.py
- misc/rge_sm.py
- unix/ffi_float.py

Do we want test directories extmod_float, misc_float, unix_float, to separate out these further, or prefix these tests with `float_` so they can be easily excluded, or don't worry about this?

pfalcon, 2015-01-08:
Thanks. Did you consider making each test file name unique (e.g. having "_float" suffix in this case), or do you thunk that dir name disambiguates it enough?

> Do we want test directories extmod_float, misc_float, unix_float, to separate out these further, or prefix these tests with float_ so they can be easily excluded, or don't worry about this?

My idea was to try to avoid float on basic/. There're more axes, e.g. different kinds of long int, etc., so I don't think it makes sense to try to split by them completely. Rather, basic tests indeed should be basic, and other more advanced tests can try to detect capabilities dynamically.

dpgeorge, 2015-01-08:
> Did you consider making each test file name unique

I did consider, but following previous pattern (eg float/string_format.py) I left them without the suffix.  Directory name should be enough (but then we should print the dir name if the test fails so we know which one it was).

> Rather, basic tests indeed should be basic

I would like even further to have a "basic basic" test dir, for which all tests pass for a uPy binary compiled with all "options" disabled (eg no unicode, no mpz, no set, ...).

pfalcon, 2015-01-29:
I didn't have chance to test it, and won't be for a while, as raw mode is non-functional currently in esp8266. So, optimistically closing this for now.

closed 2015-01-29 by pfalcon

dpgeorge, 2015-01-29:
Tested by compiling stmhal port with float disabled, and all tests in basics/ dir pass.