micropython/micropython #1048: USB VBUS detection should be a configuration option

dhylands, 2015-01-07:
Somebody else ran into this problem on the forum: http://forum.micropython.org/viewtopic.php?f=3&t=476 and I know I've helped somebody else with a similar problem for a different board.

So I thought I'd open an issue to add a new configuration called something like

MICROPY_HW_USB_VBUS_DETECT_PIN

and have it be A9 for the existing boards.

dpgeorge, 2015-01-07:
Added.  Since all STM32F4xx boards use pin A9 for this function (well, all that we currently support), then I just kept the hard coding of A9 as the detect pin.  The only change to the code is that it's not initialised if you don't define this macro.

dhylands, 2015-01-08:
I took a look at your commit, and that should certainly address the issues I've seen so far. When somebody has a pin other than A9, then we can fix the pin initialization.

dpgeorge, 2015-01-08:
Great.  If needed in the future, we can also make the DP DM pins configurable via the same method.

closed 2015-01-08 by dpgeorge