micropython/micropython #1050: Regression: RuntimeError: maximum recursion depth exceeded

btashton, 2015-01-08:
I regression was added with commit b4b10fd35
I got to this with git bisect.
You can see this running `make -c qemu-arm`

```
   text	   data	    bss	    dec	    hex	filename
  89492	    384	    608	  90484	  16174	build/flash.elf
qemu-system-arm -cpu cortex-m3 -nographic -monitor null -serial null -semihosting -kernel build/flash.elf
RuntimeError: maximum recursion depth exceeded
```
Expected output:
```
   text	   data	    bss	    dec	    hex	filename
  89548	    384	    316	  90248	  16088	build/flash.elf
qemu-system-arm -cpu cortex-m3 -nographic -monitor null -serial null -semihosting -kernel build/flash.elf
hello world!
```

dpgeorge, 2015-01-08:
Should be fixed in d2d0648ad0dd552ba8bec3425ca4ce52d3c91f62

dpgeorge, 2015-01-08:
@pfalcon now that global state is in 1 location it's all in the BSS and so now `stack_limit` has default value of 0.  Therefore one must remember to call mp_stack_set_limit.  Do you think we should make this easier by one of the following:
1. Require a parameter to mp_init, being the stack limit value.
2. Set the stack limit to a default value in mp_init, and then it can be changed by calling mp_stack_set_limit, but *after* mp_init (in all places now the calls are the other way round).

btashton, 2015-01-08:
Please consider something like pull request #1051 to catch these regressions.

pfalcon, 2015-01-08:
@dpgeorge: I considered stack limit checking to be optional feature. I got caught by this same issue when refactoring esp8266 port per https://github.com/micropython/micropython/issues/992#issuecomment-69015475 . I don't what's the best way to deal with it - maybe just not enable it by default and let individual ports deal with it.

dpgeorge, 2015-01-09:
Stack checking now disabled by default.  See 4a5895c4eba98f9c5e698d48f58cbf3682ed0db1.

closed 2015-01-09 by dpgeorge