micropython/micropython #1053: Usefulness of bare-arm

pfalcon, 2015-01-10:
Turns out, bare-arm doesn't represent runnable code. That's because of stuff like:

~~~~
size_t strlen(const char *s) {return 0;}
~~~~

So, bare-arm can answer to the question "what's the minimal size of uPy, without any library dependencies", but it doesn't answer the question "what's the minimal size of uPy" (assuming the talk about something which can really work).

So, is bare-arm useful with the info it provides? Even if it is, shouldn't we have "bare" port which actually can run (such port doesn't have to be "arm" of course).

dpgeorge, 2015-01-10:
I like bare-arm and all the void/null functions at the end of main.c because they tell you exactly what the external dependencies of uPy are.  No, it doesn't execute (and in fact has a bug in the Python script in main), but I know that it has been useful to more than 1 person in understanding how to port uPy.  Also, I use it all the time to check that code size doesn't increase too much.

Really we need a nice document overviewing the uPy structure, its dependencies, execution flow, etc.  We also need a porting guide.  But absent these, bare-arm serves a purpose.

> Even if it is, shouldn't we have "bare" port which actually can run (such port doesn't have to be "arm" of course).

Yes, I had this on my list (see eg https://github.com/micropython/micropython/issues/1045#issuecomment-69202805 which hints at this).

We could have unix/micropython_bare (just like unix/micropython_fast) which is as bare as possible but still runs, and passes a set of core tests.  We could also use qemu-arm for this purpose and make a bare but runable thumb2 executable that pases core tests (this needs work to get qemu-arm to run tests on the command line).  It wouldn't be too difficult to add both of these bare-but-runable targets, and serves a purpose to show binary size for different archs.

On a related note, I'd like to automate the selective enabling of features, and produce a table showing code size for each feature (eg unicode costs 2k).

dpgeorge, 2015-01-10:
> We could have unix/micropython_bare (just like unix/micropython_fast) which is as bare as possible but still runs, and passes a set of core tests.

FYI, I have a patch in the queue for this.

pfalcon, 2015-01-10:
me to ;-)

pfalcon, 2015-01-10:
Oops, I mean I have patch for the original idea - having a bare, but working, port.

andi6803, 2015-01-10:
Hi, I wanted to start a port for a controller from RENESA together with a student (her bachelor thesis). In the first step we started with the py borad and analysed our use cases which we want to do with uPy. In the second step we started to analyse the uPy structure and the build process. And here I have to say that this was a big challenge for us which we haven't solved until now. But you need to know that I'm not a SW professional and the student is at the beginning of her career :). So maybe it depends on our skills.
Then we tried to start with the bare-arm code and we hoped that this is the minimal code that run on the py board with objective to start the analysis from this point. But unfortunately it doesn't. I guess that this is the idea from Paul.

So from my perspective both plans of you would be very helpful for newbies who want to dive deeper into uPy.

I will definitely go on with my idea to use uPy for bring up code of new developments. And in the next time a colleague of me get the task to go on with the work from the student. Maybe then we could also contribute to this great project.

danicampora, 2015-01-13:
For me the bare-arm port was indeed very helpful when I started the port to the CC3200.

pfalcon, 2015-01-15:
Per the discussion above, we now have "minimal" port (#1062), which should be much better base to start a new port.

@andi6803 , you may want to look into it.

closed 2015-01-15 by pfalcon