micropython/micropython #1056: Consider restricting qstr length to 255 (so len storage uses 1 byte)

dpgeorge, 2015-01-11:
Interned strings (qstrs) are used for the most part for identifiers, and it's rare that a qstr will be longer than 255 bytes long.  We need to store the length of a qstr, and currently reserve 2 bytes for it, which seems wasteful.

Restricting qstrs to use only 1 byte to store their length would save ROM in the qstr table (around 200 bytes for bare-arm, and 1k for stmhal), save code size and be more efficient because you only need to access 1 byte to get the length, and also save RAM (1 byte saving for each qstr created).

Downside: it will be impossible for a user to define an identifier longer than 255 bytes.  This would raise a RuntimeError.  But who does such things?

Note: it's still possible to make arbitrarily long strings (eg "abc").

pfalcon, 2015-01-11:
> Note: it's still possible to make arbitrarily long strings (eg "abc").

Yeah, I got used to the fact that all literal strings were interned too. And only lately by a chance I saw MICROPY_ALLOC_PARSE_INTERN_STRING_LEN setting - I missed when it was added. (This may affect 12340147b0ace7c1feeaf36b618c92652bdb849f btw - @stinos, FYI).

Then, it's surely ok to limit identifier size to 255 chars.

meski, 2015-01-11:
If you're defining identifier names longer than that, you need to
re-evaluate your coding standards.

ARMWorks, 2015-01-11:
Forth used counted strings with the first byte the count. They were
invariably very handy and fast. For one thing (obvious now I'm sure) you
can reject in searches by checking the first byte. That is faster than
Boyer-Moore. +1 for me.

-- Charlie

dpgeorge, 2015-01-11:
Number of bytes used to store length is now a config option, which defaults to 1 byte.

closed 2015-01-11 by dpgeorge