micropython/micropython #1059: LED.intensity and LED(2) out of control on pyboard v3

mylettuce, 2015-01-12:
On pyboard v3, there is only LED_R1 can working on intensity mode, rigth? I check out that LED_R1 is working on Timer1 PWM mode. But stmhal code set all LED intensity is working on Timer3. So please fix it.
Follow code can set LED_R1 intensity to 1%:
    pyb.Timer(1, freq=1000)
    timer.channel(1, pyb.Timer.PWM, pin=pyb.Pin.board.LED_R1, pulse_width_percent=99)

I found LED(2) on pyboard v3 was init in AF_OD mode. If change to OUT_PP, LED(2) will be OK. Please fix it too.

dpgeorge, 2015-01-12:
Thanks for report.  LED(2) was multiplexed to USB OTG ID pin, but that's now fixed.

As for the intensity mode of LEDs, well, the reason it's not supported on PYBv3 is because Timer(1) is for general purpose use.  Timer(1) is an advanced timer and so it would be a shame to use it just for fading LEDs.  On PYBv1.0, Timer(3) is already used for the USB driver so using it also for LED fading doesn't hurt.

Do you have an idea how to make Timer(1) for LED fading, but also allow it to be used for general purpose use?

dhylands, 2015-01-12:
You could defer iniitializaion of Timer 1 until the first time that the led intensity function is called.

If we had a way of querying if the timers have been initialized or not, we could check, and if timer 1 was already initialized, just leave it, and set the led to use PWM on the appropriate channel (i.e. use the existing timer frequency).

If done generically, then intensity could work on other boards where the LEDs are assigned differently.

dpgeorge, 2015-01-12:
> If we had a way of querying if the timers have been initialized or not, we could check, and if timer 1 was already initialized, just leave it, and set the led to use PWM on the appropriate channel (i.e. use the existing timer frequency).

Yes, nice.

If the frequency of the timer is very large (eg 1MHz) will it be ok?  If the freq is very small then the LED will blink, but that's acceptable.

dhylands, 2015-01-12:
For PWM applications, the frequency mostly just affects the resolution of the PWM, and its the cycle time (when the counter rolls over and goes back to zero) is what determines the frequency the the LED sees. LEDs are often used for comms (like IrDA which can do 4 Mbps).

So I think that the LED code can do best effort and its up to the user to figure out whats appropriate if they're trying to share the timer.

dpgeorge, 2016-05-13:
LEDs can have configurable PWM support, and PYBv3 now has PWM support on LED(1) and LED(2).  See 1dc2862a837a08c6aa96fb369657be21043015ad.

closed 2016-05-13 by dpgeorge