micropython/micropython #1060: Travis CI qemu-arm tests are flaky

dpgeorge, 2015-01-12:
Sometimes the Travis CI test for qemu-arm (make -C qemu-arm test) fails for no apparent reason.  After failing, the next Travis CI run generally passes without any changes to the code.  This bug is not reproducible on my local machine.

dpgeorge, 2015-01-12:
Attempt at fix in b58da9420cdd0cca1e79be32f24088345cf66946

pfalcon, 2015-01-13:
Ack, I noticed it on weekend too.

dpgeorge, 2015-01-13:
I'm using branch travis-testing to try and fix it, without poluting master with silly commits.

doogle, 2015-01-13:
I don't know if this is related, but I'm seeing this on a Travis build atm:

```
$ make -C minimal test
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Entering directory `/home/travis/build/micropython/micropython/minimal'
mkdir -p build/genhdr
CPP ../py/qstrdefs.h
In file included from /usr/include/stdint.h:26:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.6/include/stdint.h:3,
                 from ./mpconfigport.h:1,
                 from ../py/mpconfig.h:45,
                 from ../py/qstrdefs.h:27:
/usr/include/features.h:324:26: fatal error: bits/predefs.h: No such file or directory
compilation terminated.
make: *** [build/genhdr/qstrdefs.generated.h] Error 1
make: Leaving directory `/home/travis/build/micropython/micropython/minimal'
```

`make -C minimal` builds fine on my machine. Maybe Travis are having issues? (FWIW I haven't changed any of the files mentioned).

dpgeorge, 2015-01-13:
That's a separate issue that will be fixed shortly :)  A slew of problems were introduced recently, backlog...

doogle, 2015-01-13:
Ah ok. It didn't like it was the same kind of issue but thought I'd mention it just in case.

dpgeorge, 2015-01-28:
Travis seems much better now, probably due to 2f8c409a4f441d7ef8856d2bd4ef9467204bb418.

closed 2015-01-28 by dpgeorge