micropython/micropython #1061: Roadmap for v1.4

dpgeorge, 2015-01-12:
This issue is for discussing what needs to be done to move up to release version v1.4.  The main reason for going from v1.3.x to v1.4 is that I want to merge #955, and this changes the way pyboard boots up.  Upgrading to v1.4, users will need to add 1 line to boot.py for it to have the same functionality as v1.3.x.

Here is a TODO list:

- [ ] decide on API for new HID code (see PR #955)
- [ ] incorporate new HID code
- [ ] consider incorporating #957

We can make a "v1.4-rc" branch with the changes, test them, and then set a date for merging.

dpgeorge, 2015-02-13:
I have implemented most of the USB HID stuff.  See the commits: b384bcc5de2c20d1d985e2768729578373bd73c1, d39c7aa517b722161a7dd792fafc5096d58beec8, 39ce2db181ec1e06426e2a3a371d2cd535723a17, 65af7ebdc5c53eebcec7950948abcc16d6a0ac6c, 55d6218b9a17f23754bd00d2759800a1447599ca.

It needs some testing on different machines (Windows and Mac mostly) since the USB behaviour/descriptors have changed slightly.

dpgeorge, 2015-02-13:
For the RTC/power-saving stuff, need to agree on module and function naming; see issue #1085.

pfalcon, 2015-09-13:
I wonder if status of wakeup/alarm should be updated here.

dpgeorge, 2015-09-15:
This issue is half stale.

v1.5 should probably be when stmhal uses the new hardware API #1430 and then we don't need to make backwards incompatible changes, and new stop/standby stuff can go in the machine module.

Although, maybe now is a good time to re-think version numbers.  The uPy core is getting mature but stmhal will still have lots of new features, so these 2 components could do with separate version numbering.  This is how WiPy will work (a version for the wipy firmware, different to uPy core version).

pfalcon, 2015-10-18:
> v1.5 should probably be when stmhal uses the new hardware API #1430 and then we don't need to make backwards incompatible changes, and new stop/standby stuff can go in the machine module.

Well, such combination of events may deserve even 2.0 imho. Generally, while we're way behind CPython in version numbers, I don't think we need to spare them too tightly.

> Although, maybe now is a good time to re-think version numbers. The uPy core is getting mature but stmhal will still have lots of new features, so these 2 components could do with separate version numbering.

Now, with all the fixes done recently, I agree that it's getting mature. But it's not yet there to call for such split versioning, especially that it takes more effort. YMMV.

dpgeorge, 2016-05-06:
We are now on version 1.8 and the items in the checklist in the first post of this issue are either outdated or implemented.

closed 2016-05-06 by dpgeorge