micropython/micropython #1064: Converting minimum long long value to string results in incorrect output

doogle, 2015-01-13:
When mp is configured to use 'long long' ints (`MICROPY_LONGINT_IMPL == MICROPY_LONGINT_IMPL_LONGLONG`), the following code:
```
print(-0x7fffffffffffffff-1)
```
results in the string:
```
-'..--).0-*(+,))+(0(
```
rather than:
```
-9223372036854775808
```

This is due to `mp_obj_int_formatted()` trying to negate the number, which fails in this case. The subsequent conversion to a string then results in invalid characters due to (I think) signedness issues.

pfalcon, 2015-01-16:
Well, such issues are boring to fix. The core of it is indeed in the fact that after negating negative value, it should be treated as unsigned.

doogle, 2015-01-16:
I found it while doing the float to int stuff, so I was mainly recording it for posterity. It's definitely minor and, as you said, not the most exciting of coding challenges :-)

I'll have at it once I've got around to adding some tests to my float to int pull request.

dpgeorge, 2017-04-11:
Fixed by e31fbd9b41315963aaca732e7585fc4cb10ccbcd.

closed 2017-04-11 by dpgeorge