micropython/micropython #1075: OSError: 5 with i2c

ice3, 2015-01-20:
Hello, 

I have ported micropython on FEZ-CERB40.
When I use i2c on it a OSERROR: 5 appears.

Here is the faulty code:
``` python
class Azotec():
    def __init__(self):
        # config :
        ## see the module datasheet for more infos
        ## values optained wit azotec debug tool on Windows
        b = bytearray([9, 10, 7, 10, 100, 100, 10, 5, 7])
        while not i2c.is_ready(0x74):  # necessary ? 
            pass
        i2c.mem_write(b, 0x74, 0x11, timeout=100)
```

Here is the result:
``` shell
Traceback (most recent call last):
  File "main.py", line 142, in <module>
  File "main.py", line 85, in __init__
OSError: 5
```

reproduction using REPL:
``` python
Micro Python v1.3.7-169-ge12217c on 2015-01-08; FEZ-CERB40 with STM32F405RGT
Type "help()" for more information.
>>> print(i2c.scan())
[116]
>>> b = bytearray([9, 10, 7, 10, 100, 100, 10, 5, 7])
>>> i2c.mem_write(b, 116, 0x11)              
>>> i2c.mem_write(b, 116, 0x11)
>>> i2c.mem_write(b, 116, 0x11)
>>> i2c.mem_write(b, 116, 0x11)
>>> i2c.mem_write(b, 116, 0x11)
>>> i2c.mem_write(b, 116, 0x11)
>>> i2c.mem_write(b, 116, 0x11)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 5
```

The i2c seem's to work (worked 7 times but not the 8th) but fail randomly. 

I had the same problem on STM32f4discovery, so I don't think the issue is CERB specific.

dhylands, 2015-01-20:
One reason that will cause the error is if the device that you're trying to talk to doesn't generate an ACK on the address, then OSError: 5 will be returned. Failing to return an ACK on any other byte causes a timeout (which I think is OSError: 116)

You'd really need a logic analyzer to confirm this.

dpgeorge, 2015-01-21:
Yes, @dhylands is correct: I think the only way OSError:5 occures is due to an "acknowledge failure".  It could be a problem with the I2C device, or the wiring.

ice3, 2015-01-23:
Yes there was a problem with the way I used the device, thanks. 
The device sets a ready (RDY) channel when it is ready, if you read while not ready you will encounter a OSError: 5.
Glad it's not a micropython issue :)

For people curious about how to handle the rdy state, we can't use callback to read the I2C, so we poll : 
``` python
def write_i2c(mess):
    while rdy.value() == 0:
        pass
    return  i2c.mem_write(mess, 0x74, 0x11)

def read_i2c():
    while rdy.value() == 0:
        pass
    return  i2c.mem_read(36, 0x74, 0x01)
```

closed 2015-01-23 by ice3

dpgeorge, 2015-01-24:
@ice3 Okay, good that it works.  BTW, you may want to replace "pass" with "pyb.wfi()" to reduce power consumption.  "pyb.wfi()" will sleep, waiting for the next interrupt (wait-for-interrupt).  Interrupts should be enabled, then this instruction will wait at most 1ms, since that's the time of the next systick.  It could wake earlier, if some other interrupt occurs.  If reading/writing to the i2c device is not time critical (ie can wait 1ms) then pyb.wfi() will allow you to reduce power consumption in this busy while loop.