micropython/micropython #1091: Import this, results in incorrect output

danicampora, 2015-01-27:
Just for fun, I tried putting`this.py` and doing`import this` on the pyboard. To my surprise, I was only able to see the last line of the text (The Zen of Python). I debugged the issue, and what happens is that the decoded lines have no LF characters, only CR, so, in the terminal every new line overwrites the previous one. Is it because we need a better implementation of `stdout_tx_strn_cooked` that can also handle line endings with CR only, or is there something else going wrong? Off course, this is not high priority, but I thought it was worth mentioning...

Here is the `this.py` that I used:
```python
s = """Gur Mra bs Clguba, ol Gvz Crgref

Ornhgvshy vf orggre guna htyl.
Rkcyvpvg vf orggre guna vzcyvpvg.
Fvzcyr vf orggre guna pbzcyrk.
Pbzcyrk vf orggre guna pbzcyvpngrq.
Syng vf orggre guna arfgrq.
Fcnefr vf orggre guna qrafr.
Ernqnovyvgl pbhagf.
Fcrpvny pnfrf nera'g fcrpvny rabhtu gb oernx gur ehyrf.
Nygubhtu cenpgvpnyvgl orngf chevgl.
Reebef fubhyq arire cnff fvyragyl.
Hayrff rkcyvpvgyl fvyraprq.
Va gur snpr bs nzovthvgl, ershfr gur grzcgngvba gb thrff.
Gurer fubhyq or bar-- naq cersrenoyl bayl bar --boivbhf jnl gb qb vg.
Nygubhtu gung jnl znl abg or boivbhf ng svefg hayrff lbh'er Qhgpu.
Abj vf orggre guna arire.
Nygubhtu arire vf bsgra orggre guna *evtug* abj.
Vs gur vzcyrzragngvba vf uneq gb rkcynva, vg'f n onq vqrn.
Vs gur vzcyrzragngvba vf rnfl gb rkcynva, vg znl or n tbbq vqrn.
Anzrfcnprf ner bar ubaxvat terng vqrn -- yrg'f qb zber bs gubfr!"""

d = {}
for c in (65, 97):
    for i in range(26):
        d[chr(i+c)] = chr((i+13) % 26 + c)

print ("".join([d.get(c, c) for c in s]))
```

danicampora, 2015-01-27:
For what is worth, here is a slightly modified version of `uart_tx_strn_cooked()` that seems to work fine:

```c
void uart_tx_strn_cooked(pyb_uart_obj_t *self, const char *str, uint len) {
    for (const char *top = str + len; str < top; str++) {
        if (*str == '\n') {
            uart_tx_char(self, '\r');
        }
        else if ((*str == '\r') && (*(str + 1) != '\n')) {
            uart_tx_char(self, '\n');
        }
        uart_tx_char(self, *str);
    }
}
```

pfalcon, 2015-01-28:
Nak, cannot reproduce.

Why think there's something wrong with *output*, not *input*?

danicampora, 2015-01-28:
Did you try with the pyboard? Well, that's the standard ´this.py´ that ships with Python3. If it works fine with CPython, I think it should also work well with MicroPython.

pfalcon, 2015-01-28:
I did try the pyboard. But in your report, there're enough details missing, so providing them may shed some light why it doesn't work for you.

danicampora, 2015-01-28:
What details do you need? I created a file called "this.py" (with the contents that you see in the original post) inside "/flash/lib". Then on the pyboard REPL, I typed:
```python
import this
```
And instead of getting the same ouput you could get with CPython, which is:
```
The Zen of Python, by Tim Peters

Beautiful is better than ugly.
Explicit is better than implicit.
Simple is better than complex.
Complex is better than complicated.
Flat is better than nested.
Sparse is better than dense.
Readability counts.
Special cases aren't special enough to break the rules.
Although practicality beats purity.
Errors should never pass silently.
Unless explicitly silenced.
In the face of ambiguity, refuse the temptation to guess.
There should be one-- and preferably only one --obvious way to do it.
Although that way may not be obvious at first unless you're Dutch.
Now is better than never.
Although never is often better than *right* now.
If the implementation is hard to explain, it's a bad idea.
If the implementation is easy to explain, it may be a good idea.
Namespaces are one honking great idea -- let's do more of those!
```

I only got the last line with some extra characters at the end that belong to the previous lines.

I thought that most people here would know about that easter egg, and that no more details were needed.

danicampora, 2015-01-28:
Besides, your response of: 
````
Nak, cannot reproduce
```

It's also not giving much details of what you tried. I simply gave an easy to reproduce example which in my opinion shows that CPython handles line endings a bit different than we do. You are welcome to prove me wrong, I have no interest in being always right.
Is it that hard to be nice from time to time? There's no need to have a Linus Torvalds attitude all day long...

pfalcon, 2015-01-28:
> What details do you need?

Just the same as almost any page, a hit for "how to report bugs" suggests ;-). Summing up, they usually suggest be precise and detailed, and try to look at issue "from outside" - all to avoid doubt and guesswork "on the other end".  

> It's also not giving much details of what you tried.

That's true, even though I did thorough testing - I tested unix build, unix minimal build, and pyboard. I just didn't tell anyone, and now from outside it looks like I have Linus Torvalds attitude... (is that trying to provide quick response to issues instead of just ignoring "bad" reports? ;-) ).

Ok, being nice line link: http://www.chiark.greenend.org.uk/~sgtatham/bugs.html . And as you're a developer yourself, I guess have a reason to hope that you would try to make fellow developer's lives easier by trying to provide detailed reports, so sorry for the rant.

pfalcon, 2015-01-28:
Ok, as enough info was not provided, let's reduce to guesswork (which means I may be wrong, and it sucks). So: you're doing these tests on Windows. Python you take that file from is also installed on Windows. So, the files it installs have CRLF line-endings. A file with such line-ending is what you push to pyboard, and that's what gives you those "CR". As I hinted, it's input, not output issue then.

Nothing that happens: 1) on unix; 2) when pasting code at the interactive prompt, including on Windows (well, unless terminal settings are also, well, "peculiar"). Note that for a small code snippet, testing via pasting at interactive prompt is natural way, so someone could really need a bunch of thinking and trying to figure out what's wrong on your side, without your providing system details and step-by-step instructions.

So, is the above explains the issue you see?

danicampora, 2015-01-28:
Ok, agreed that I didn't said that I did it from windows, but I did say that the problem was when trying:
```python
import this
```
which already tells you not to paste the code in the interactive prompt. The original title of the issue was already explicit about it.

Anyway, thanks for the explanation of the issue.

pfalcon, 2015-01-28:
So, if that's it, please update title to be even more precise, and it can be discussed how to deal with it.

danicampora, 2015-01-28:
I updated the title and also the original post. I believe it's precise enough now.

danicampora, 2015-01-29:
Ok, I tried another test. I have a file with the following contents:
```
s = """line ending test line 0
line ending test line 1
line ending test line 2
last line, last line, last line"""

print (s)
```
if the file is saved under Linux (with the name of "le_test"), so that it has LF line endings, we get this on the pyboard:
```python
>>> import le_test
line ending test line 0

line ending test line 1
line ending test line 2
last line, last line, last line
```
But if the file is saved on Windows (with CRLF line endings), then the result is:
```python
>>> import le_test
last line, last line, last line
```

So, like @pfalcon pointed out, the problem is caused by the input having CRLF line endings. Then the question is, should we deal with this? or simply document that MicroPython requires Unix line endings to work properly and forget about it.

dpgeorge, 2015-01-29:
But I would think that it should fail differently: if the line endings are CRLF (`\r\n`) then when it prints the string it just dumps all chars through the stdout_tx_strn_cooked function on pyboard, which converts CR to CRLF.  Thus the line endings should be CRLFLF, which should move the cursor to the far left and down 2 lines.

danicampora, 2015-01-29:
Yes, that's what I would think as well, but at the moment it reaches `stdout_tx_strn_cooked()`, the LF characters are gone.

dpgeorge, 2015-01-29:
Ok.  That's because the lexer converts CRLF to CR within strings.  So that's the crux of the problem.  The lexer should either not convert, or convert to LF.  Also, there is a related bug: making a string that's not terminated with closing quotes but has a CRLF at the end of the line won't lex correctly.

danicampora, 2015-01-29:
I think it should convert to LF in order to play nice with `stdout_tx_strn_cooked()`, unless there's a better reason not to.

pfalcon, 2015-01-29:
The simplest testcase is:

~~~~
print(repr("""foo
bar"""))
~~~~

put in a file with CRLF lineendings. Output expected for Windows users is `'foo\nbar'`. However, what Unix users should expect? One idea is that everything in triple-quoted strings should be literal. However, the same file, run on Ubuntu with official 3.4 package, and with 3.4.2 built from source leads to the same `'foo\nbar'` output. That means that "universal newline module" is now default for Python, and we should follow the behavior.

dpgeorge, 2015-01-29:
Yes, it indeed seems that CPython3 converts CRLF to LF in triple quoted strings.

dpgeorge, 2015-01-30:
Fixed in 32bade19d952c23c2fd499050abb17cb80dc7559.  Thanks @danicampora for finding a very subtle bug!

closed 2015-01-30 by dpgeorge