micropython/micropython #1092: 'bytes()' function seems to be broken

blmorris, 2015-01-28:
'bytes()' is incorrectly encoding values greater than 127 (0x7F)
I haven't figured out where the relevant code is, but I think that it showed up in the last two weeks or so. 
I found it after pulling the latest code last night - before that I had last pulled the micropyton code on Jan 11, and it wasn't broken then.

The incorrect output appears on both unix and stmhal:
```
Micro Python v1.3.6-339-g0ecd598 on 2015-01-27; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> bytes([0xAA, 0x55])
b'\xc2\xaaU'
>>> bytes([127])
b'\x7f'
>>> bytes([128])
b'\xc2\x80'
>>> bytes([127, 128])
b'\x7f\xc2\x80'
>>> bytes([255])
b'\xc3\xbf'
```

On CPython:
```
Python 3.4.2 (default, Oct 15 2014, 21:24:41) 
[GCC 4.2.1 Compatible Apple LLVM 4.2 (clang-425.0.28)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> bytes([0xAA, 0x55])
b'\xaaU'
>>> bytes([127])
b'\x7f'
>>> bytes([128])
b'\x80'
>>> bytes([127, 128])
b'\x7f\x80'
>>> bytes([255])
b'\xff'
```

Off topic - the uPy version that shows up when the REPL starts has not been tracking the version numbers of the main tree when I pull new code changes - is this a minor bug or am I doing something wrong?

pfalcon, 2015-01-28:
Ack. Will have a look.

pfalcon, 2015-01-28:
> v1.3.6

OTOH, the above is too old . Do you practice git fetch --tags from time to time?

blmorris, 2015-01-28:
The source tree is current; I confirmed the bug still exists pulling the latest changes 1.5 hours ago.
However, ```git fetch --tags``` is the thing that I didn't know about. Thanks!

blmorris, 2015-01-28:
Just ran ```git fetch --tags``` to confirm that this bug really is in the current source:

```
Micro Python v1.3.9-15-g98e3a64 on 2015-01-28; linux version
>>> bytes([255])
b'\xc3\xbf'
>>> bytes([0x80])
b'\xc2\x80'
>>> 
```
FWIW, my use case is constructing a binary format data frame to transmit to an application running on the host via USB:
``` python
port = pyb.USB_VCP()
sync = bytes([0xAA, 0x55])

def checksum(buf):
    sum = 0
    for element in buf:
        sum += element
    return 256 - sum % 256

def frame_transmit(message):
    header = sync + bytes([len(message), checksum(message)])
    frame = header + message
    return port.send(frame)

…

retval = frame_transmit(msg_data)
```
```msg_data``` gets constructed correctly, as it doesn't use ```bytes()```, but ```sync``` and sometimes ```checksum``` have values > 0x7F

While tracking this down I noticed one other inconsistency with CPython - when passed a value > 255, CPython throws an exception:
```
>>> bytes([256])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: bytes must be in range(0, 256)
```
Currently, ```bytes([256])``` returns ```b'\xc4\x80'```, before this bug it returned ```b'\x00'``` which is reasonable but inconsistent with CPython. These cases don't appear to be tested in the testsuite.

pfalcon, 2015-01-28:
First reported issue is fixed in bbd9251bac5bf7e2c2055a8b17172b46e7ff8cd0.

pfalcon, 2015-01-28:
For 2nd issue, I recommend submitting a separate bug, which would have lower priority (essentially, if it's not reported that not catching values >255 breaks existing 3rd-party Python code, it's not worth fixing it for a while, as it will only spend precious bytes).

blmorris, 2015-01-28:
Thanks, that fixed it for me.

> For 2nd issue, I recommend submitting a separate bug, which would have lower priority

Will do.

closed 2015-01-28 by blmorris