micropython/micropython #1096: REPL prints AT+GCAP command

slush0, 2015-02-01:
After every startup of REPL (both screen or minicom), board sends strange output to the terminal:

> AT+GCAP
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> NameError: name 'AT' is not defined
> >>> AT+GCAP
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> NameError: name 'AT' is not defined
> >>> ~x~~x~

It appeared in original older firmware (built sometimes in 10/2014), today I reflashed to nightly build, same issue. Board factory reset does not help as well.

slush0, 2015-02-01:
I'm ashamed, it took me three months to realize that it's actually an issue of Ubuntu/Mint modem-manager and not an pyboard issue: http://www.linuxquestions.org/questions/linux-software-2/screen-something-keeps-typing-at-gcap-to-my-terminal-4175496244/

Stopping modem-manager solved the problem. At least I updated pyboard to latest firmware ;-).

Anyway, pyboard & micropython is awesome!

closed 2015-02-01 by slush0

dhylands, 2015-02-02:
@slush0 You can create a custom udev entry for the pyboard with the following (I call mine /etc/udev/rules.d/49-micropython.rules)
```
# f055:9800 - MicroPython board
ATTRS{idVendor}=="f055", ATTRS{idProduct}=="9800", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="f055", ATTRS{idProduct}=="9800", ENV{MTP_NO_PROBE}="1"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="f055", ATTRS{idProduct}=="9800", ATTRS{manufacturer}=="Micro Python", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="f055", ATTRS{idProduct}=="9800", ATTRS{manufacturer}=="Micro Python", MODE:="0666"
```

The first line (with ID_MM_DEVICE_IGNORE) tells modem manager to ignore this serial port.