micropython/micropython #1100: Inline assembler error reporting

peterhinch, 2015-02-03:
It would be a major improvement if the Syntax Error message emitted a line number. Currently the only way to locate a defective line seems to be by a process of trial and error, commenting out lines until the message is no longer output.

peterhinch, 2015-02-04:
Failure to report a line number applies only to the problem I reported in #1099 and that other syntax errors do produce a correct line number.

dpgeorge, 2015-02-04:
Not all inline-asm errors will quote a line number, so let's keep this issue open until the situation improves.

dpgeorge, 2015-02-13:
Error reporting for inline assembler is now much improved by 8dfbd2d5894c98bd7b48e3e69bec0216d06ebb47.  Line numbers and function name are given, and it's now a real exception that can be caught.

closed 2015-02-13 by dpgeorge