micropython/micropython #1103: qstr_from_strn assertion failed

slush0, 2015-02-07:
Hi, I'm trying to get "ecdsa" package working (pure-python implementation of elliptic curves).

I hit such assertion error and I suspect it's something "low level" in micropython:

>>> import ecdsa
micropython: ../py/qstr.c:151: qstr_from_strn: Assertion `len < (1 << (8 * (1)))' failed.
Aborted

I'm running Micro Python v1.3.9-30-g471b2a8 on 2015-02-02; linux version

pfalcon, 2015-02-07:
Please try to make a short self-contained testcase to help debugging this.

slush0, 2015-02-07:
I have no idea where it crashed because of no stack trace is provided. I'm trying to isolate the issue but no success so far.

slush0, 2015-02-07:
Ha, I just found it!

$ micropython 
Micro Python v1.3.9-41-g8785645 on 2015-02-07; linux version
>>> Msg = 0x84ce72aa8699df436059f052ac51b6398d2511e49631bcb7e71f89c499b9ee425dfbc13a5f6d408471b054f2655617cbbaf7937b7c80cd8865cf02c8487d30d2b0fbd8b2c4e102e16d828374bbc47b93852f212d5043c3ea720f086178ff798cc4f63f787b9c2e419efa033e7644ea7936f54462dc21a6c4580725f7f0e7d1
micropython: ../py/qstr.c:151: qstr_from_strn: Assertion `len < (1 << (8 * (1)))' failed.
Aborted

dpgeorge, 2015-02-07:
The issue is that this literal is longer than 255 characters and is being interned.  (Big ints start their life as a qstr, then get parsed at runtime.)  Proper solution is to parse the literal in the parser and create a big-int object there-and-then.  Issue with this approach is that the parser can now raise exceptions.

slush0, 2015-02-07:
Is it something which might be fixed eventually? Such big numbers are quite typical for crypto applications and right now it blocks porting of ECDSA to micropython.

dpgeorge, 2015-02-07:
Well, I'm working on it right now :)

danicampora, 2015-02-07:
Sounds like this solution will also bring some performance improvements, so it's a win win, isn't it?

dpgeorge, 2015-02-07:
Yes, it will bring RAM and speed optimisations.  Issue #722 is closely related.

dpgeorge, 2015-02-08:
Fixed.

closed 2015-02-08 by dpgeorge

slush0, 2015-02-08:
Works for me. Awesome!