micropython/micropython #1104: Confusing line number reporting for exceptions in loops

pfalcon, 2015-02-07:
Testcase:

~~~~
l = ["-foo", "-bar"]

i = 0
while l[i][0] == "-":
    print("iter")
    i += 1
~~~~

So, error happens in line 4, however uPy backtrace:

~~~~
$ micropython test.py 
iter
iter
Traceback (most recent call last):
  File "test.py", line 6, in <module>
IndexError: list index out of range
~~~~

In real-world use, the loop body is much bigger, and line number refers not to last statement in loop body, but to statement after the loop, which makes it even more confusing.

dpgeorge, 2015-02-15:
Problem comes about because of the following 2 facts:
1. While loops are optimised in the standard way by putting the condition test after the body (eliminates an unconditional jump).
2. Line number encoding assumes a monotonic map between bytecode offset and line number.

To fix this issue we need to change one of the above.

dpgeorge, 2022-06-24:
I wouldn't want to de-optimise while loops (option 1) above, and reworking line number mapping is difficult.  A third option would be to only apply the while optimisation at opt-level 3 and above, where line numbers are anyway disabled.  But that would increase code size of the firmware, and most users don't enable optimisations so wouldn't benefit from this.

I don't think this is that much of an issue anyway, and the difference is tracked in the cpydiff tests.

closed 2022-06-24 by dpgeorge