micropython/micropython #1105: TypeError: type 'module' is not an acceptable base type

slush0, 2015-02-07:
I just hit this difference between python3 and upython:

$ micropython
Micro Python v1.3.9-41-g8785645 on 2015-02-07; linux version
>>> from types import ModuleType
>>> class foo(ModuleType): pass
TypeError: type 'module' is not an acceptable base type

but

$ python3
Python 3.3.2+ (default, Feb 28 2014, 00:52:16) [GCC 4.8.1] on linux
>>> from types import ModuleType
>>> class foo(ModuleType): pass
>>>

I can probably work-around it, but I'm just curious if it's expected or not.

dpgeorge, 2015-02-08:
Is subclassing a module a common thing?  It's not hard to make your above code work, but then your use of the `foo` type will probably break down some other place.  How is the `foo` type used?

slush0, 2015-02-08:
You're right that it's a part of code which probably won't work either in micropython because of metaprogramming. I wanted to get python-ecdsa work out of the box, but now I see I'll need to simplify some parts of the code.

slush0, 2015-02-08:
For the record, it's about this package: https://github.com/warner/python-ecdsa/blob/master/ecdsa/six.py which achieve compatibility for python2/python3. It should be straightforward to simplify it to work just for micropython.

closed 2015-02-08 by slush0

dpgeorge, 2015-02-08:
six.py will push the limits of uPy :)  We can get it to run, but probably at too much expense.  It is anyway a Py 2/3 compatibility layer, so can probably be rewritten to provide compatibility for uPy.

slush0, 2015-02-08:
I didn't know six.py before, but it looks like general purpose script for writing applications which are compatible for py2 and py3. Maybe if we (well - you, because I'm avoiding C as much as possible to prevent brain damage :-) can get six.py running on micropython, much more applications would run on micropython out of the box...

Anyway, I'm forking six.py and trying to get it run by removing all that metadata programming...

dpgeorge, 2015-02-08:
If you're serious about getting ecdsa running on uPy, then I can make a branch with changes to get the features you need.  Then we can see if they are suitable to pull back into the master branch.

slush0, 2015-02-08:
Fine, give me some time, I'll hack the compatibility code and if there'll be a chance that it will be doable, then we can make the code "nice" by implementing that metadata stuff...

mdupuy, 2017-07-26:
2017 calling. Actually running in to a need for six.py in micropython. Did you ever get anywhere with your port?

brady-aiello, 2018-07-30:
2018 calling. Did @slush0 port six.py to micropython? I'm also trying to get python-ecdsa running on micropython.