micropython/micropython #1106: Porting to ARM Cortex M0?

joseangeljimenez, 2015-02-08:
Hi all,

I am trying to make a rough estimation about the effort it may take me to port Micropython to a more limited platform than the current one: ARM Cortex M0, 256KB Flash, 32KB RAM.

Thanks for your comments and suggestions,
Jose Angel

P.S.: I am quite experienced with several MCU and DSP architectures. Also with Python (desktop). I have very little experience with ARM Cortex.

dpgeorge, 2015-02-08:
It should work "out of the box"; try the directory minimal/ as a starting point.  You'll have a basic REPL, but will need to implement board-support functions yourself (or copy from stmhal/).

dpgeorge, 2015-02-09:
If you need further help, please ask/discuss on the forum: http://forum.micropython.org

closed 2015-02-09 by dpgeorge

joseangeljimenez, 2015-02-09:
Thanks a lot for your nice help!

jjtara, 2016-04-04:
@joseangeljimenez , did you started the migration? I'm interested as well

joseangeljimenez, 2016-04-20:
@jjtara Hi Juanjo, I am not there yet, I am currently involved in another project!