micropython/micropython #1107: to_bytes expects int, fail strangely on float

slush0, 2015-02-08:
Hi, when float is passed to to_bytes, strange exception is generated.

>>> int('1').to_bytes(1.0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 70015440351760 bytes

 I understand micropython does not validate all inputs to keep codebase tiny and agile, so if this is under the radar, just close the issue...

dpgeorge, 2015-02-08:
It's assuming that the argument is an int and extracting the value without checking the type.  It's a 1 line fix.

slush0, 2015-02-08:
Of course it was my fault and the fix is int(1.0), I just reported that error handling. Python3 returns
>>> int('1').to_bytes(1.0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: integer argument expected, got float

instead of MemoryError about allocation of 70TB RAM.

dpgeorge, 2017-06-15:
Fixed by e269cabe3ed8bed1b7181359febb686edbb748ae

closed 2017-06-15 by dpgeorge