micropython/micropython #1109: TypeError: unsupported types

slush0, 2015-02-09:
> class _bytes(bytes):
>     pass
> bytes() + _bytes()
TypeError: unsupported types for __add__: 'bytes', '_bytes'

This passes in python3 without problem.

Goal: I'm missing some extended functionality in built-in types (like bytes.translate), so I wanted to patch them by pure python implementation in overloaded class. Of course I can implement them in separate object instead, but it's missing a beauty of transparency.

pfalcon, 2015-02-09:
Subclassing native types is complicated matter, and only subset of that is supported. Adding more support for it isn't easy and unambiguously good: first of all, it's hard (existing code is already complicated); it takes noticeable amount of code, but very few people and apps need it, so it wastes code space, and potentially makes all other code a bit slower. So, anyone who really needs those features are welcome to contribute them (#ifdef'ed) ;-).

You can try to workaround the issue above by defining explicitly `__add__` on bytes_ (to explicitly call `bytes.__add__`). That certainly should be supported (but may have some bugs, but we at least want to fix them with some priority).

pfalcon, 2015-02-09:
>  first of all, it's hard (existing code is already complicated)

Some reference which has more references inside: https://github.com/micropython/micropython/issues/624

dpgeorge, 2015-02-09:
> You can try to workaround the issue above by defining explicitly __add__ on bytes_ (to explicitly call bytes.__add__). That certainly should be supported

Nope, `bytes` does not have `__add__` attribute in uPy :(  In fact, no builtin types have special method names corresponding to unary or binary ops.

Fixing this issue "properly" might be the right way to go, and actually rather easy: just need to provide support for the buffer protocol in user defined types (eg define the special method name `__buffer__`) and then delegate to a builtin subclass if the user didn't define it.

dpgeorge, 2015-02-09:
Ok, in 48244044a2b434406f95b4df76835262cda09dba I added support for subclassed native types to delegate to the native's get_buffer function (if it exists).  Apart from resolving this issue, this is generally useful for uPy which uses the buffer protocol in a lot of places to avoid memory allocation.

Next would be to define a special method name for the user to hook into the buffer protocol.  I propose `__buffer__` which acts as follows:

```python
class MyBuffer:
    def __init__(self):
        self.internal_buf = bytearray(10)

    # flags = 1 for read, 2 for write, 3 for read-write
    def __buffer__(self, flags):
        if flags & 2:
            return None # don't support writing
        else:
            return self.internal_buf # delegate to object that has buffer protocol
```

It is easy and cheap to implement this feature.  Thoughts?

blmorris, 2015-02-09:
I'm mostly revealing my ignorance here, but several months ago I inquired on the forum about the possibility of creating a buffer object as an interface to read data out of a file. This would allow me to open a binary file containing data to be transmitted over I2C without first reading the entire file contents into a 'bytes' object.
I don't actually need this feature for my application; the maximum size of the binary data to program my I2C device is 8.5KB. This data object can fit in RAM and only needs to be loaded and transmitted once on startup. It would be easy to imagine a situation where the entire contents of a file would not fit into available RAM, and it could be useful to create a buffer interface to stream data into or out of a file. 
Would this feature enable that usage?

dpgeorge, 2015-02-10:
> This would allow me to open a binary file containing data to be transmitted over I2C without first reading the entire file contents into a 'bytes' object.

I think you can already do this:
```python
b=bytearray(8000)
f=open('file', 'rb')
f.seek(1000) # skip 1000 bytes
f.readinto(b) # read next 8000 into b
f.close()
```

slush0, 2015-02-10:
Thanks for the __buffer__ support, I'll check it out!

pfalcon, 2015-02-10:
> It is easy and cheap to implement this feature. Thoughts?

My quick thought is that we're not going to have standard Python feature like `__add__`, but going to have non-standard `__buffer__`. Somehow I feel there's some trick in resolving the reported issue in such way (maybe it's because I didn't face it, I wonder if extensions I propose also look not immediately clear from outside).

pfalcon, 2015-03-15:
> In fact, no builtin types have special method names corresponding to unary or binary ops.

That's not exactly true - there's ``set.__contains__``, ``dict.__getitem__``, etc. Those were added because I saw stdlib using them. So, those can be added for all objects and methods, the question is why - they would be very rarely used.

Coming back to the original issue reported by @slush0 (lack of bytes.translate), he already found the easiest way to work that around - just implement separate translate() function and patch couple of places (sic, it's another rarely used feature) were it was called. 2nd choice would be to implement bytes.translate() for interested people to enable in their builds.

Working the issues discussed here by adding non-standard ``__buffer__`` is rather roundabout way (that method may be useful, but first good usecase should be shown).

And btw, even in CPython, there're limits up to which subclassing builtin types work: https://mail.python.org/pipermail/python-dev/2015-February/138184.html . So, uPy isn't much different, we just placing even more restrictions (to keep it small).

projectgus, 2024-08-28:
Closing this out as the original reported code works (I think for a long time now). The general case is complex, but there's a documented set of CPython incompatibilities now. If someone runs into a problem it's probably better they open an issue for that specific case.

closed 2024-08-28 by projectgus