micropython/micropython #1110: Failed conversion to bytes

slush0, 2015-02-09:
>>> print(14953042807679334000 & 0xff)
>>> print(ord(bytes([14953042807679334000 & 0xff])))

Both lines should print "112", as Python3 does. Micropython prints "64" for second line, which is mistake.

I hit this by implementing sha512 which deals with large integers.

slush0, 2015-02-09:
Ugly workaround:

>>> x = int(float(14953042807679334000 & 0xff))
>>> print(ord(bytes([x])))

blmorris, 2015-02-09:
I can confirm the bug, but found something even more odd:
```
Micro Python v1.3.9-49-g2863153 on 2015-02-08; linux version
>>> bytes([14953042807679334000 & 0xff])
b'\x00'
>>> bytes([14953042807679334000 & 0xff])
b'\x90'
>>> bytes([14953042807679334000 & 0xff])
b'`'
>>> bytes([14953042807679334000 & 0xff])
b'0'
>>> bytes([14953042807679334000 & 0xff])
b' '
>>> bytes([14953042807679334000 & 0xff])
b'\xd0'
>>> bytes([14953042807679334000 & 0xff])
b'\xa0'
>>> bytes([14953042807679334000 & 0xff])
b'p'
>>> bytes([14953042807679334000 & 0xff])
b'`'
>>> bytes([14953042807679334000 & 0xff])
b'\x10'
```
Basically returning a different value every time.
BTW, when embedding a code snippet if you wrap it in a triple backquote  \`\`\`  then `>>>` won't appear as indenting.

pfalcon, 2015-02-09:
Fixed in 194117a066fa0af6cbe712223195e4ff02037298.

closed 2015-02-09 by pfalcon

slush0, 2015-02-09:
Awesome, works for me.