micropython/micropython #1111: Inline assembler function causes crash

peterhinch, 2015-02-09:
Use of registers beyond r7 seems problematic. The following causes a crash
```
@micropython.asm_thumb
def foo(r0):
    mov(r9, r0)

foo(0)
```

dpgeorge, 2015-02-09:
r9 is a callee save register, so you must somehow save/restore it in your asm function.

The lower registers are already saved for you for convenience.

So, this needs documenting :)

peterhinch, 2015-02-10:
I also had a repeatable crash with the same code and R8. R10 seems to work. It would indeed be good to know which we can safely use - I'm using r0-r7 and need another :)

Alternatively, if registers beyond r7 are out of bounds it would be good to know how to push and pop regsters onto the stack. (I'm a noob regarding ARM assembler). Perhaps I should pursue this in the forum.

dpgeorge, 2015-02-10:
Thumb2 calling conventions are: callee save regs r4-r11 and r13=sp.  r4-r7 are saved for you.  push/pop are not yet implemented.  Note that you can always use the "data" directive to create unsupported instructions: `data(bytesize, data, ...)`; eg `data(2, 0x1234, 0x4567)`.

peterhinch, 2015-02-10:
TFT. I've raised a query on the forum as I guess that's a more appropriate place for coding advice.

peterhinch, 2015-02-12:
I've documented this in the Wiki so I guess this can be closed.

closed 2015-02-12 by peterhinch

dpgeorge, 2015-02-13:
push and pop are now implemented.  Use as:
```python
push({r8, r9})
pop({r8, r9})
```

pfalcon, 2015-02-13:
A set? Looks funky.

dpgeorge, 2015-02-13:
I wanted it to be as close to ARM assembly as possible.  This syntax then allows for the more general instructions:
```python
ldmia(r0, {r1, r2})
stmdb(r0, {r1, r2)}
```

pfalcon, 2015-02-13:
Yeah, I see. Though might be interesting to think how to be ready to support other archs, and then having simpler single-reg push/pops might be useful. But yes, it's arch-dependent anyway, so other asms would probably need their own syntax.

EmmBee, 2015-02-13:
Does "push({r8, r9})" mean push r8, followed by r9 ?
If so then R9 is tos so shouldnt it be "pop({r9, r8})" ? that way you know that R9 came off the stack first...
Just an old lurker testing how sharp his razor is at splitting hairs;)

dpgeorge, 2015-02-13:
The register list is encoded as a bit pattern: 16 bits, each one set if that register should be pushed/popped.  So the order is fixed and it doesn't matter which way you write it.  When the CPU executes them, push/stm do the opposite order to pop/ldm.  I think the higher numbered registers are pushed first, ie take the higher memory address.