micropython/micropython #1112: [hardware]

cloudformdesign, 2015-02-11:
Have you seen the Sparkcore project? You guys are using almost the same chip,  except their photon comes with a broad com wireless module built into the same silicon for only $10! (that is both the wifi and the silicon) 

It would be awesome to have their wifi and hardware stack and python together.  You guys should talk! 

https://store.spark.io

danicampora, 2015-02-11:
Well, soon we will be launching a WiFi module running MicroPython on the
CC3200. Dimensions will be very similar to the ones of the sparkcore.

cloudformdesign, 2015-02-11:
Wow, that is pretty awesome chip. I didn't realize the next gen of the CC3xxx was a SoC

The only thing I would caution against is that Spark had difficulty with the drivers for the CC devices. I don't know if it was a problem with their coding or the chip, but it was a serious difficulty -- causing such things as the whole system freezing. They advertise that the Broadcom they use on the chip is in products like *Nest* as well as Wifi Routers, so it is supposed to be very reliable and robust.

Just some food for thought. It is practically the same architecture (from my understanding) as your target platform and may offer better reliability. Either way, love the project!

danicampora, 2015-02-11:
Hey Garret,

Thanks for your comments!!
I do know the CC3000 had some reliability problems, but as far as I have been able to exprience, the CC3200 is rock solid, no weird issues at all. Time will tell, of course...

Ivoah, 2015-02-24:
Would it be possible to get micropython running on the Photon? It uses a STM32F205.

cloudformdesign, 2015-02-24:
Awesome! I hope it works out well. I am pretty sure that this project will revolutionize the way I program micro-controllers.

closed 2015-02-24 by cloudformdesign