micropython/micropython #1113: timer.deinit() doesn't stop the timer

rowpie, 2015-02-12:
The documentation for **timer.deinit()** suggested that it will stop the timer and disable the [timer] peripheral.   I don't think it does.
See forum topic: http://forum.micropython.org/viewtopic.php?f=2&t=544

I've poked about in stmhal/timer.c at the **pyb_timer_deinit()** call from where **HAL_TIM_Base_DeInit()** is called.   I wondered if a call to **HAL_TIM_Base_Stop()** is required but my expermiments with that didn't stop the timer either.

dpgeorge, 2015-02-13:
Should be fixed now (I tested your example with the PWM and it stops it).  Issue was that the HAL deinit function (and stop function) did nothing because there were channels that were enabled, and it wants all channels disabled before disabling the timer.  Easy fix was just to reset the timer control registers completely.

closed 2015-02-13 by dpgeorge

rowpie, 2015-02-13:
Great stuff. I see the direct register resets.   Works as I expect now.
Many many thanks.