micropython/micropython #1114: Relative import error

Ivoah, 2015-02-13:
I have a directory structure like this:

test/
|___ __init__.py
\___ x
    |___ __init__.py
    \___ y.py

test/__init__.py has:
```python
from .x import *
print('init')
```

test/x/__init__.py is:
```python
from .y import *
print('x')
```

and test/x/y.py is:
```python
print('y')
```

The output from CPython is:
```
y
x
init
```

While micropython throws an error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test/__init__.py", line 1, in <module>
  File "test/x/__init__.py", line 1, in <module>
ImportError: no module named 'test.y'
```

Micropython tries to import test.y, while it should be importing test.x.y

dpgeorge, 2015-02-13:
@pfalcon will you have time to look at this or should I have a go?

pfalcon, 2015-02-15:
I'll see if I can look into it during today's flights. If not, I'd treat with lower priority (workaround is known).

pfalcon, 2015-02-16:
Fixed in 9e6c82960e38963e4392ad3180811d10657519b3

closed 2015-02-16 by pfalcon