micropython/micropython #1115: Pyboard DAC does not deinit correctly

dpgeorge, 2015-02-13:
See the forum for details: http://forum.micropython.org/viewtopic.php?f=6&t=497&start=20#p3063

clacktronics, 2015-02-13:
Thanks!  Ive tested it with the following code in v1.3.8 im not sure if there is issues with the way ive written it ( not sure if I should define dac variable in the function for both pins? ) but found out this.

```python

import pyb
from time import sleep

def pwmWrite(pin,pwm):
	buf = bytearray(100) 
	dac = pyb.DAC(pin) 
	for i in range(len(buf)):
		if i < pwm:
			buf[i] = 255
		else:
			buf[i] = 0
	dac.write_timed(buf,1000 * len(buf), mode=pyb.DAC.CIRCULAR)
	
	
for i in range(0,101):
	
	pwmWrite('X5',i)
	pwmWrite('X6',i)
	sleep(.1)

```
Run as is USB plugged in = 4 led crash, X5 acts weird ( perhaps code written wrong? )
X5 commented out USB plugged in = 4 led crash
X5 commented out no USB = runs fine
X6 commented out USB plugged in = 4 led crash
X6 commented out no USB = runs fine

Then I thought maybe its because I keep defining the DAC so I separated it out into a global variable.

```python
import pyb
from time import sleep

dac = (pyb.DAC('X5'),pyb.DAC('X6'))

def pwmWrite(pin,pwm):
	global dac
	buf = bytearray(100) 
	for i in range(len(buf)):
		if i < pwm:
			buf[i] = 255
		else:
			buf[i] = 0
	dac[pin].write_timed(buf,1000 * len(buf), mode=pyb.DAC.CIRCULAR)
	
	
for i in range(0,101):
	
	pwmWrite(0,i)
	pwmWrite(1,i)
	sleep(.01)

```

This runs fine! but sometimes X5 disappears at last step 1 out of 5 times if the USB is in.

Does this suggest its perhaps something to do with the USB? and is it really good bashing the pyb.DAC function over and over?

dpgeorge, 2015-02-13:
A very simple fix, and it would make your code much more efficient, is to just allocate a single bytearray for the DAC buffer, and overwrite it each time you want to change the pulse width.  The DAC will keep writing out the data using the same buffer.  Eg:
```python
class DACPWM:
    def __init__(self, pin, resolution):
        self.dac = pyb.DAC(pin)
        self.buf = bytearray(resolution)
        self.dac.write_timed(self.buf, 1000 * len(self.buf), mode=pyb.DAC.CIRCULAR)

    def pulse_width_percent(self, pct):
        b = self.buf # cache buf variable for speed
        pwm = pct * len(b) // 100
        for i in range(len(self.buf)):
            b[i] = 255 if i < pwm else 0
```

clacktronics, 2015-02-13:
OK so that never crashes! is it because the code is inefficient then?
thanks for showing me those efficiency tricks!

dpgeorge, 2015-02-13:
It's not because it's inefficient that it crashes.  A crash should be fixed, and it's good that you exposed a bug!  But calling write_timed on the DAC sets up DMA transfers which is not fast, especially if you do it many times per second.

dpgeorge, 2017-04-11:
The code posted here no longer crashes the pyboard, so closing the issue.

closed 2017-04-11 by dpgeorge