micropython/micropython #1117: non-intern strings not works for thumb native code

deepgully, 2015-02-13:
MICROPY_ALLOC_PARSE_INTERN_STRING_LEN set to 10
```python

@micropython.native
def test():
    print("0123456789")     # works
    print("01234567890")    # not works for thumb native code, works for x64 

test()
```

dpgeorge, 2015-02-13:
Works for me with latest build on x86, x64 and pyboard.

deepgully, 2015-02-14:
in function `asm_thumb_mov_reg_i32_aligned`, the i32 will be stored and a BN op will emitted before to jump over the value, what's the reason?
there is the native code: http://www2.onlinedisassembler.com/odaweb/t2YN43/0, 0x412f1740 is the addr of non-intern string object

It works after I disable the BN op and the i32 value.
```c
// i32 is stored as a full word in the code, and aligned to machine-word boundary
// TODO this is very inefficient, improve it!
void asm_thumb_mov_reg_i32_aligned(asm_thumb_t *as, uint reg_dest, int i32) {
    // align on machine-word + 2
    if ((as->code_offset & 3) == 0) {
        asm_thumb_op16(as, ASM_THUMB_OP_NOP);
    }
    // jump over the i32 value (instruction prefect adds 4 to PC)
    // asm_thumb_op16(as, OP_B_N(0));
    // store i32 on machine-word aligned boundary
    //asm_thumb_data(as, 4, i32);
    // do the actual load of the i32 value
    asm_thumb_mov_reg_i32_optimised(as, reg_dest, i32);
}
```
the new native code: http://www2.onlinedisassembler.com/odaweb/gQrTlp/0

deepgully, 2015-02-14:
P.S. my CPU is Cortex-R4 series

dpgeorge, 2015-02-15:
The 4-byte pointer must be stored in the code so the GC can trace it, and it must be aligned on a machine word.  But, as you suggest, it does look like it should be OP_B_N(2), since the prefetch adds 2 to the PC not 4.  Now fixed.

Thanks for finding this bug!

closed 2015-02-15 by dpgeorge