micropython/micropython #1122: ure Library -  MemoryError

clacktronics, 2015-02-16:
Not sure if this is a feature of it being micro or a bug, but if I nest brackets in a regular expression then try to call the upper group I get the following.

```
ure.match('(([0-9]*)([a-z]*)\d*)','1234hello567').group(1)
```
This crashes it ( four LED's hard reboot )

and in other cases I get this issue

```
>>> ure.match('([0-9]*)(([a-z]*)([0-9]*))','1234hello567').group(2)
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 4294966749 bytes
```

danicampora, 2015-02-16:
It is most likely a bug. I am not qualified to say so ;-), but sounds like it's something similar to https://github.com/micropython/micropython/issues/1107

dpgeorge, 2015-02-21:
Should now be fixed.

closed 2015-02-21 by dpgeorge

clacktronics, 2015-03-04:
Thanks for the fix! although just to let you know the memory error is fixed but doing this for example 

```python

Micro Python v1.3.10-143-g2a68c2c on 2015-03-04; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import
>>> ure.match('(#)? *([a-z]*)','test').group(1)
```

Still crashes it making the 4 led's flash, is this something to do with the USB?

dpgeorge, 2015-03-04:
Well, you found another bug!  Issue was that a group with no match wasn't handled correctly.  Now fixed.