micropython/micropython #1123: HID messages touchscreen problem

ice3, 2015-02-17:
Hello, 

I want to "create" a touch screen (more exactly only the touch part) with micropython. 
I use a custom HID descriptor. 

Here is boot.py :
``` python

import pyb
#pyb.main('main.py') # main script to run after this one
hid_desc = bytes([
    0x05, 0x0D,        # Usage Page (Digitizer)
    0x09, 0x04,        # Usage (Touch Screen)
    0xA1, 0x01,        # Collection (Application)
    0x85, 0x03,        #   Report ID (3)
    0x05, 0x0D,        #   Usage Page (Digitizer)
    0x09, 0x54,        #   Usage (Contact count)
    0x95, 0x01,        #   Report Count (1)
    0x75, 0x08,        #   Report Size (8)
    0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x06, 0x00, 0xFF,  #   Usage Page (Vendor Defined 0xFF00)
    0x75, 0x02,        #   Report Size (2)
    0x09, 0x01,        #   Usage (Button 1 ?)
    0x81, 0x01,        #   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x75, 0x0E,        #   Report Size (14)
    0x09, 0x02,        #   Usage (0x02)
    0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x0D,        #   Usage Page (Digitizer)
    0x09, 0x22,        #   Usage (Finger)
    0xA1, 0x02,        #   Collection (Logical)
    0x05, 0x01,        #     Usage Page (Generic Desktop Ctrls)
    0x16, 0x00, 0x00,  #     Logical Minimum (0)
    0x26, 0xFF, 0x07,  #     Logical Maximum (2047)
    0x75, 0x0B,        #     Report Size (11)
    0x09, 0x30,        #     Usage (X)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x0D,        #     Usage Page (Digitizer)
    0x25, 0x1F,        #     Logical Maximum (31)
    0x75, 0x05,        #     Report Size (5)
    0x09, 0x48,        #     Usage (Width)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x01,        #     Usage Page (Generic Desktop Ctrls)
    0x16, 0x00, 0x00,  #     Logical Minimum (0)
    0x26, 0xFF, 0x07,  #     Logical Maximum (2047)
    0x75, 0x0B,        #     Report Size (11)
    0x09, 0x31,        #     Usage (Y)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x0D,        #     Usage Page (Digitizer)
    0x25, 0x1F,        #     Logical Maximum (31)
    0x75, 0x05,        #     Report Size (5)
    0x09, 0x49,        #     Usage (Height)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x25, 0xFF,        #     Logical Maximum (255)
    0x75, 0x08,        #     Report Size (8)
    0x09, 0x51,        #     Usage (Contact Identifier)
    0x95, 0x01,        #     Report Count (1)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x30,        #     Usage (Tip Pressure)
    0x75, 0x05,        #     Report Size (5)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x42,        #     Usage (Tip Switch)
    0x15, 0x00,        #     Logical Minimum (0)
    0x25, 0x01,        #     Logical Maximum (1)
    0x75, 0x01,        #     Report Size (1)
    0x95, 0x01,        #     Report Count (1)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x47,        #     Usage (Confidence)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x32,        #     Usage (In Range)
    0x81, 0x02,        #     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,              #   End Collection
    0xC0,              # End Collection
    ])

report_len = 12  # bytes
pyb.usb_mode("VCP+HID", pid=0x1234, hid=(0, 0, report_len, hid_desc))

```

It a custom touch HID descriptor, detecting only one finger. 
When I send a HID message from the code or the interpretor, the message is sent correctly. 

But dmesg starts shouting error messages :
``` shell

Feb 17 14:10:28 hoofse ModemManager[807]: <info>  (tty/ttyACM0): released by modem /sys/devices/pci0000:00/0000:00:1a.1/usb4/4-2
Feb 17 14:10:35 hoofse kernel: [14864.020056] usb 4-2: new full-speed USB device number 19 using uhci_hcd
Feb 17 14:10:35 hoofse kernel: [14864.206726] usb 4-2: New USB device found, idVendor=f055, idProduct=1234
Feb 17 14:10:35 hoofse kernel: [14864.206730] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 17 14:10:35 hoofse kernel: [14864.206732] usb 4-2: Product: Pyboard Virtual Comm Port in FS Mode
Feb 17 14:10:35 hoofse kernel: [14864.206734] usb 4-2: Manufacturer: Micro Python
Feb 17 14:10:35 hoofse kernel: [14864.206736] usb 4-2: SerialNumber: 000000000011
Feb 17 14:10:35 hoofse mtp-probe: checking bus 4, device 19: "/sys/devices/pci0000:00/0000:00:1a.1/usb4/4-2"
Feb 17 14:10:35 hoofse kernel: [14864.216780] input: Micro Python Pyboard Virtual Comm Port in FS Mode as /devices/pci0000:00/0000:00:1a.1/usb4/4-2/4-2:1.0/input/input31
Feb 17 14:10:35 hoofse kernel: [14864.216981] hid-multitouch 0003:F055:1234.0015: input,hidraw3: USB HID v1.11 Device [Micro Python Pyboard Virtual Comm Port in FS Mode] on usb-0000:00:1a.1-2/input0
Feb 17 14:10:35 hoofse kernel: [14864.217044] cdc_acm 4-2:1.1: This device cannot do calls on its own. It is not a modem.
Feb 17 14:10:35 hoofse kernel: [14864.217063] cdc_acm 4-2:1.1: ttyACM0: USB ACM device
Feb 17 14:10:35 hoofse mtp-probe: bus: 4, device: 19 was not an MTP device
Feb 17 14:10:51 hoofse ModemManager[807]: <info>  Creating modem with plugin 'Generic' and '1' ports
Feb 17 14:10:51 hoofse ModemManager[807]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
Feb 17 14:10:51 hoofse ModemManager[807]: <warn>  Couldn\'t create modem for device at '/sys/devices/pci0000:00/0000:00:1a.1/usb4/4-2': Failed to find primary AT port
Feb 17 14:11:11 hoofse kernel: [14899.710730] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.718731] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.726731] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.734729] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.742725] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.750729] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.758725] usb 4-2: input irq status -75 received
Feb 17 14:11:11 hoofse kernel: [14899.766724] usb 4-2: input irq status -75 received
...
```

From my investigation, the ```input irq status -75 received```  comes from [this file](https://github.com/torvalds/linux/blob/0349678ccd74d16c1f2bb58ecafec13ef7110e36/drivers/hid/usbhid/hid-core.c)

I'm not sure if these messages could come from malformed HID messages by the pyboard (ie if my descriptor is wrong or the pyboard doesn't form them correctly).

ice3, 2015-02-17:
I think the problem comes from the pyboard. 
I have tried this with my report descriptor : 
``` python
>>> hid.send(bytearray(range(10)))
>>> hid.send(bytearray(range(11)))
```

The first line go smoothly but the second triggers the ``` input irq status -75 received```

dpgeorge, 2015-02-17:
You might need to rate limit your HID reports; ie wait until you are allowed to send the next one:
```python
import select
while True:
    rd, wr, ex = select.select([], [hid], [])
    if wr:
        hid.send(...)
```

ice3, 2015-02-17:
Hum, I see. 
Nevertheless the select is blocking. 

```python
az = Azotec()

while True :
    az.xy = read_i2c()
    t = az.hid_touches()
    print(t)
    print("before select")
    rd, wr, ex = select.select([], [hid], [])
    print("after select")
    if wr:
        if az.touches:
            print(".", end="")
            hid.send(t)
```

Works but after sending the first HID message block at "before select" and the error message continue to appear in dmesg

ice3, 2015-04-29:
The problem does not come from micropython. 
I have reproduced it on an Arduino Leonardo. 

It happens when the HID report (the message send, ie, the bytearray) is too long according the descriptor. 
I managed to make the HID descriptor work on the Arduino. I will try tomorrow with micropython, I keep you in touch with the results. 

I will also give the HID descriptor because it is quite a pain to find correct information.

dpgeorge, 2018-06-20:
> It happens when the HID report (the message send, ie, the bytearray) is too long according the descriptor.

According to this statement, it looks like this issue was resolved.

closed 2018-06-20 by dpgeorge