micropython/micropython #1125: Standarizing make_new() methods

danicampora, 2015-02-18:
I noticed, that the parameters format of the `make_new()` methods is not standard for classes within the pyb module. For example, many of them accept keyword arguments (Pin, I2C, SPI), but some others don't (ADC is one example). Also, `socket.socket()` doesn't accept kargs the way is implemented now (stmhal, cc3200), while according to CPython it should.
We should decide if we will always accept keyword arguments for creating objects, and at the same time follow CPython's pattern. Or we could choose not support them at all (to reduce code size for example). Either way we go, it should be consistent across all modules.
Thoughts?

pfalcon, 2015-02-19:
> Either way we go, it should be consistent across all modules.

Why it should be "consistent across all modules"? It's per-function matter, and rule of thumb is well-known: "If a function can accept a lot of args, but each invocation realistically will pass a few, then using kwargs makes sense". For uPy, there's additional rule of thumb: "Avoid using kwargs, as it take more space and are slightly slower".

Due to second one, we don't accept kwargs everyone CPython does, and I guess that's good, and we should keep not accepting until *several* good reports (and good reports are those adhering to "Django criteria" - it was spoken few times already, so I proceeded to capture it in https://github.com/micropython/micropython/wiki/ContributorGuidelines#django-criteria for reference (to remind it's all RFC, and subject to discussion in #1007)).

So, I think we're good in general (as long as it's documented).

There may be some specific cases which can be made more consistent, feel free to raise them. But if the proposal is to add keywords where they weren't before, I'd say it's low priority, as it will only eat up bytes, and we have bigger inconsistencies, which hopefully should be resolved with higher priority.

danicampora, 2015-02-19:
@pfalcon Thanks for the info. My proposal wasn't to add keywords everywhere, just to make behavior consistent. But if the decision was to keep it as per-module needs, then it is OK. Now that I know the "rule of thumb", then it's clear how to proceed.
Your comments make it all pretty clear, so I'll close this one. Thanks!

danicampora, 2015-02-19:
@pfalcon by the way, nice work on the [Contributor Guidelines](https://github.com/micropython/micropython/wiki/ContributorGuidelines)!

closed 2015-02-19 by danicampora

dpgeorge, 2015-02-19:
For the record, classes like Pin and UART take keywords to their init functions because the functions are complicated and keywords make it very clear what the call is doing.

The `timeout` arg of send/recv methods was chosen to be keyword-only again for clarity; it's pretty confusing what the numbers do if you just write `i2c.send(1, 2, 3)`.

If 1 arg of a function is parsed as a keyword, then it's simpler to allow them all to be keywords and use the helper function `mp_arg_parse_all`.  Otherwise, if the function is simple enough, don't use keywords to save space.

danicampora, 2015-02-19:
Ok, understood. Thanks!