micropython/micropython #1126: Inline assembler error reporting

peterhinch, 2015-02-21:
Code instances where the line number of the error is not reported
```
@micropython.asm_thumb
def foo(r0):
    push({r0)

#Traceback (most recent call last):
#  File "<stdin>", line 1, in <module>
#SyntaxError: invalid syntax
'''
The following error produces different outcomes depending on context. In some cases it produces the following outcome:
```
# code
   add(r2, r0)
# more code
>>> import asm

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: unsupported Thumb instruction 'add' with 2 arguments
```
But in the following case the assembler accepts it with no error reported:
```
def bar(r0):
    add(r2, r0)
```

dpgeorge, 2015-02-24:
There are 4 versions of add (and sub):

- add(rd, rm, rn) : rd = rm + rn
- add(rd, rm, i3) : rd = rm + i3 (i3 up to 3 bits immediate value)
- add(rd, i8) : rd += i8 (i8 up to 8 bits immediate value)
- add(rd, rm) : rd += rm

uPy currently supports only the first 3 versions.

dpgeorge, 2015-02-24:
Are you sure about the first issue with the unmatched brace?  That is anyway a parser error and has nothing to do with the inline assembler.  I've tested your code and for me it reports that the error is on line 3.

peterhinch, 2015-02-24:
Regarding add() I appreciate that add(r2, r0) is unsupported and the line is in error: my issue is with its reporting. In some cases the line number is 1. In other cases the code is accepted - I haven't checked the outcome but presumably it's poor.

Regarding the unmatched brace I've installed today's build (24th) and I'm still getting the Line 1 output as above. Very odd.

dpgeorge, 2015-02-24:
Ok, I see.  Can you please post exact code that fails (or minimal code that fails) and exactly how you run it (do you use tools/pyboard.py for example?).

peterhinch, 2015-02-26:
The Micropython board is connected to the PC with USB and I run it using screen /dev/ttyACM0. The build is that of 24th Feb (pybv10-2015-02-24-v1.3.10-87-g4fddbe5.dfu)
The file, run from a Micro SD card, is as follows:
```
[adminpete@tapir]: /mnt/upython
$ cat errors.py 

@micropython.asm_thumb
def foo(r0):
    push({r0)

#print(foo(1))

```
Output:
```
>>> 
PYB: sync filesystems
PYB: soft reboot
Used to run nrf24l01 slave
Micro Python v1.3.10-87-g4fddbe5 on 2015-02-24; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import errors
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: invalid syntax
>>> 
```
Sorry for the slow response.

dpgeorge, 2015-02-26:
Thank you.  I confirm the bug: it happens when you import a script with a syntax error.

dpgeorge, 2015-02-26:
This is a regression, and fa3b8951452ca13a807c0b6a090e0385507113db is the offending commit.

dpgeorge, 2015-02-27:
The problem with the syntax error line number reporting is fixed.

Do you have an example where the add instruction gives an error?

peterhinch, 2015-02-27:
The time when add(r2, r0) produced an error it was in the middle of a long program. My attempts to produce a sensible example yielded the outcome where the faulty instruction was ignored by the compiler. I'll try to come up with an example that can sensibly be posted.

peterhinch, 2015-02-27:
I flashed the board with today's build and put the error into the following code:
```python
@micropython.asm_thumb
def root(r0):
    push({r2, r3, r4})
    clz(r4, r0)
    neg(r4, r4)
    add(r4, 32) # no. of bits in result
    mov(r3, 1)
    asr(r4, r3)
    mov(r2, r0)
    asr(r2, r4) # r2 first estimate of result
    label(REPEAT)
    udiv(r4, r0, r2)
    add(r4, r2) # ERROR. Should be add(r4, r4, r2)
    asr(r4, r3)
    cmp(r4, r2)
    beq(DONE)
    mov(r2, r4)
    b(REPEAT)
    label(DONE)
    mov(r0, r2)
    pop({r2, r3, r4})
```
The outcome was as follows:
```
Micro Python v1.3.10-104-gd155fec on 2015-02-27; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import isqrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: unsupported Thumb instruction 'add' with 2 arguments
```
i.e. it correctly reports the error but omits the line number.

peterhinch, 2015-02-28:
The assembler also omits to report duplicate labels. I'd suggest there's little need to output line numbers, simply a message on the lines of "multiple definitions of label <name>". At present such user stupidity is typically rewarded with a crash.

dpgeorge, 2015-03-03:
The problem with add not reporting the correct line number was fixed in the above-referenced commit.  Your build d155fec was the commit just before the fix :)  I have checked that the above code with "root" gives an error message with the correct line number.

dpgeorge, 2015-03-03:
Duplicate labels now give an error; see 9c5cabb502b59e5cc4354c8ca66be3d4a3e03bf9.

closed 2015-03-03 by dpgeorge