micropython/micropython #1128: REPL hangs on Timer object creation

nadrimajstor, 2015-02-22:
After pressing RESET button and typing:
``` py
Micro Python v1.3.10 on 2015-02-13; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> n = 3
>>> def countdown(timer):
...     global n
...     if n <= 0:
...         print("Launch!")
...         timer.deinit()
...     else:
...         print(n)
...         n -= 1
... 
>>> tim = pyb.Timer(4, freq=1, callback=countdown)
```
At this moment REPL hangs and I need to press RESET to recover. :cry: 

Probably related issue:
``` py
Micro Python v1.3.10 on 2015-02-13; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> n = 3
>>> def countdown(timer):
...     global n
...     if n <= 0:
...         print("Launch!")
...         timer.deinit()
...     else:
...         print(n)
...         n -= 1
... 
>>> tim = pyb.Timer(4)
>>> tim.callback(countdown)
>>> tim.init(freq=1)
>>> # Nothing happens :|
```
Nevertheless it does work if I continue and provide both kwargs:
``` py
>>> tim.init(freq=1, callback=countdown)
3
>>> 2
1
Launch!
```
However, if I continue with:
``` py
>>> tim.deinit()
>>> tim.callback(countdown)
>>> Launch!
```
I did not expect after issuing `deinit()` to have triggered callbacks without explicitly calling `init` with `freq` kwarg. :confused: 
And after issuing a soft reset
``` py
PYB: sync filesystems
PYB: soft reboot
Micro Python v1.3.10 on 2015-02-13; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> tim = pyb.Timer(4, callback=print("I did not set freq option"))
I did not set freq option
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: must specify either freq, or prescaler and period
>>> tim = pyb.Timer(4)
>>> tim.callback(print("I did not set freq option"))
I did not set freq option
>>> 
```
It certainly comes as a surprise that callback is still kicking without explicit `freq`

dhylands, 2015-02-23:
I recoded it as:
```
tim = pyb.Timer(4)
tim.init(freq=1)
tim.callback(countdown)
```
and then it works fine. Any callback set prior to calling init will be wiped out by callig init (which is why your example didn't work).

I'll see if I can figure out why passing callback= to init isn't working.

dhylands, 2015-02-23:
I found the issue, and created PR #1131 to fix the problem.

dpgeorge, 2015-02-23:
The main bug raised by this issue is now fixed: setting the callback in the constructor works as expected.

But the OP shows some other "bugs" with the Timer class: calling any of the methods that reconfigure the timer (such as freq and callback) before the timer is initialised (either using the constructor or using init method) will have undefined behaviour.  Fix is either to raise an exception when freq/callback is called without the timer being initialised, or to just document it saying that these reconfiguration methods only work correctly after the timer has been initialised.

dhylands, 2015-02-23:
I like the idea of raising an exception, but this adds code that is only needed for malformed python.

We should probably be consistent across peripherals (timer, uart, spi, i2c, files, etc).

I consider calling stuff between deinit and init to be similar to calling read/write between a close and an open.

nadrimajstor, 2015-02-23:
For what it's worth, at the moment, it's kinda hard to work through Timer examples as callbacks are triggered when unexpected... It's difficult to issue `tim.callback(None)` while the REPL is constantly filling up with new lines. :blush:  hw reset is not a option - remember (*put your favorite computer platform without a fdd or hdd*) how you felt when your marvelous previous typing was suddenly all gone. :smile_cat:
IMHO, raising an exception or just "nothing happens" are more intuitive way towards need user corrective actions.
Nevertheless, thank you for your time and patience. (Please do note if my novice poking at things is, at the moment, counter productive.)

dhylands, 2015-02-24:
For testing, I dislike typing, so I tend to create the file on my host, copy it to the USB Mass Storage and then import thee file on the REPL.

So for this bug, I created a file called countdown.py which had the following in it:
```
import pyb
import micropython

n = 3
def countdown(timer):
    global n
    if n <= 0:
        print("Launch!")
        timer.deinit()
    else:
        print(n)
        n -= 1

micropython.alloc_emergency_exception_buf(100)
tim = pyb.Timer(4, freq=1, callback=countdown)
```

I copied it to /media/dhylands/PYBFLASH and then waited about 5 seconds and on the REPL I do:
```
>>> <Press Control-D>
PYB: sync filesystems
PYB: soft reboot
Micro Python v1.3.10-87-g4fddbe5 on 2015-02-23; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import countdown
3
>>> 2
1
Launch!
```

When I want to make changes, I edit the file on the host and repeat the process.