micropython/micropython #1129: ADC and DAC xxx_timed() methods can't be used concurrently

ul5255, 2015-02-22:
DAC.write_timed(..., mode=pyb.DAC.CIRCULAR) and ADC.read_timed(...)
can't be used concurrently. The call to ADC.read_timed() stops the signal
generated by DAC.write_timed(). For details see http://forum.micropython.org/viewtopic.php?f=3&t=550&start=10#p3159

ul5255, 2015-02-22:
I'd suggest to allocate one timer to the ADC(s) and a *separate* timer to the DAC(s).

dpgeorge, 2015-02-22:
> I'd suggest to allocate one timer to the ADC(s) and a separate timer to the DAC(s).

Yes, that's a reasonable thing to do.  Could even allow to pass a timer object in as a kwarg to select the timer, with default being tim6.

peterhinch, 2015-03-06:
That has the advantage of generality: for example you might want to allocate a timer to each DAC instance to enable them to output asynchronously.

ul5255, 2015-04-27:
Should *I* should go ahead and implement this? If yes: What is the proper way of doing that? I have another pull request (https://github.com/micropython/micropython/pull/1130) which is untouched since 22-FEB-2015 even though I thought I followed the instructions for pull requests. Implementing different timers for DAC and ADC is certainly more involved.

dpgeorge, 2015-04-27:
@ul5255 yes this is something you can try to implement.  I didn't yet because it's difficult :)  Especially since DMA restricts which timers work with which peripherals.  And also I want to make ADC.read_timed work with DMA.

Please feel free to start thinking/hacking on these ideas.  Eg a good start is to simply list all the timers and DMA channels for DAC and ADC and see which ones work with each other.

dpgeorge, 2015-07-21:
DAC now allows you to specify a different timer for write_timed, so both ADC and DAC can work together (ADC still uses Timer(6) always).

Would be nice to be able to specify timer for ADC similar to DAC, so leaving this issue open until that is implemented.

bmentink, 2015-07-21:
Can you also specify a different timer for each DAC channel? Is the current change checked in, I would like to try it.

dpgeorge, 2015-07-21:
Yes and yes. See the most recent commits.

dpgeorge, 2015-07-22:
ADC.read_timed now allows to specify a timer to use for the triggering.  For example:
```python
adc = pyb.ADC(pyb.Pin("X19"))
buf = bytearray(100)
tim = pyb.Timer(2, freq=100)
adc.read_timed(buf, tim)
```
You can use any timer to set the read frequency.  You can then use a different timer (or even the same one if you want the same frequency) for DAC.write_timed.

closed 2015-07-22 by dpgeorge

bmentink, 2015-07-22:
Terrific! .... now all we need is DMA support for the Timer and it gill be game-set-and-match ;)