micropython/micropython #1132: Inline assembler: asm_thumb_bcc_n: branch does not fit in 9 bits

peterhinch, 2015-02-23:
I'm seeing this in a program with about 350 lines of assembler when I add extra lines. It's occurring as a runtime error, output before a crash. Shouldn't Python detect this at compile time? The code isn't doing anything to preclude this - there are no computed jumps and all branches are to labels.

dpgeorge, 2015-02-23:
In this case you have exceeded the capabilities of the inline assembler :)  Some work needs to be done to support wide jumps.

dpgeorge, 2015-02-23:
Work around is to replace the long branches with a branch to a closer instruction which then branches to the final destination.  Or, neater, to negate the condition code and jump over an unconditional branch:
```
    bge(NOT_DEST)
    b(DEST)
    label(NOT_DEST)
```

peterhinch, 2015-02-23:
I'm puzzled that the error occurs at runtime and isn't picked up by the compiler/assembler. However I've fixed my code without the extra debugging code which was provoking the problem so my personal issue has gone away. It's up to you whether to leave this issue open but a compile time error message would be helpful.

dpgeorge, 2015-02-25:
Fixed.

closed 2015-02-25 by dpgeorge