micropython/micropython #1140: Problem using const() with stm module

blmorris, 2015-02-27:
Trying to use the stm module to get I2S working in MicroPython, I encountered a strange error.
My application uses the I2S2 peripheral in full-duplex mode, which requires initializing both the I2S2 and I2S2ext instances; I2S2 is just a specific configuration of SPI2 so the address is the same, but the address offset for I2S2ext is not defined in the stm module.
I attempted to use `const()` to define the address offsets:
```
Micro Python v1.3.10-109-g0bb9713 on 2015-02-27; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import stm
>>> I2S2 = const(stm.SPI2)
>>> I2S2ext = const(I2S2 - 0x400)
Traceback (most recent call last):
  File "<stdin>", line 1, in  ¡=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ±=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿ­=ÿµ
SyntaxError: constant must be an integer
>>> I2S2ext = const(stm.SPI2 - 0x400)
```
Note that the last line works without an error.
Does this have anything to do with the stm constants being treated as signed 31-bit integers? I have notice that to get the constants to display properly (as 31-bit unsigned integers) I need to mask the 32nd bit:
```
>>> I2S2
-1073727488
>>> hex(I2S2)
'-0x3fffc800'
>>> hex(I2S2 & 0x7fffffff) 
'0x40003800'
````
The last line prints the correct memory offset for the SPI2 / I2S2 peripheral.

dhylands, 2015-02-27:
Interesting. My pyboard crashes (fatal error) if I do:
```
x1 = const(32)
x2 = const(x1 + 1)
```

blmorris, 2015-02-27:
@dhylands - I can confirm that as well.

dpgeorge, 2015-03-01:
Error message reporting and hard crash now fixed.  Problem was that the compilation scope was not set correctly before compiling the constants, so error messages did not work.

The other "issue" is that you can't define constants at the REPL (well, you can, but they won't act as optimised constants).  The const feature only works within an single compilation unit (eg a file).  At the REPL each line is an individual compilation unit so consts are not remembered from one line to the next.

dpgeorge, 2015-03-01:
stm now has I2S2EXT and I2S3EXT constants.

closed 2015-03-01 by dpgeorge