micropython/micropython #1142: Inline assembler data statement: suggested enhancement

peterhinch, 2015-03-01:
The data statement enables data to be included with code. Unfortunately it doesn't seem possible to access it without explicitly manipulating the program counter. Attempts to assign labels to registers produce an error. For example it would be good to be able to write something like this (to access an element from a data array):
```python
d1 = const(12)
d2 = const(13)

@micropython.asm_thumb
def bar(r0):
    b(START)

    label(MYDATA)
    data(4, d1, d2)

    label(START)
    movwt(r1, MYDATA)
    add(r0, r0, r0) # Convert index to byte offset
    add(r0, r0, r0)
    add(r1, r1, r0) # add offset to data address
    ldr(r0, [r1, 0])
```
But this fails with "unsupported Thumb instruction 'movwt' with 2 arguments". The code can be made to work as follows, but it seems inelegant compared to using a label.
```python
d0 = const(1234567)
d1 = const(987654)
d2 = const(3*d1)

@micropython.asm_thumb
def getdata(r0):
    push({pc})
    b(START)
    data(4, d0, d1, d2)
    label(START)
    pop({r2})
    add(r2, 2)          # skip the b() instruction
    add(r0, r0, r0)
    add(r0, r0, r0)     # Convert array offset to bytes
    add(r2, r2, r0)     # Add to adjusted address
    ldr(r0, [r2, 0])    # get the requested data
```

methoxid, 2017-10-04:
mowvt and mow do not seem to work with named constants 

`from micropython import const
STpinnr = const(5)
STpinreg=const(stm.GPIOB)
@micropython.asm_thumb
def adcw():
   movwt(r6, STpinreg) # #def toggleB5(): get the GPIOB address in r1
   movw(r5, 1 << STpinnr) # get the bit mask for pin number << N

Traceback (most recent call last):
  File "<stdin>", line 3, in adcw
SyntaxError: unsupported Thumb instruction 'movwt' with 2 arguments
`
This works however:
`@micropython.asm_thumb
def adcw():
   movwt(r6, stm.GPIOB) # #def toggleB5(): get the GPIOB address in r1
   movw(r5, 1 << 5) # get the bit mask for pin number << N
`

dpgeorge, 2017-10-04:
@methoxid if you use paste mode (ctrl-E) then your example works ok.

jonnor, 2024-07-02:
Is this issue still relevant? @peterhinch

peterhinch, 2024-07-03:
I'm not sure. The point was that using the `data` statement is difficult. Conventionally you would create a label at the start of a data array and reference the data through the label. The PR made this possible, although the code is arguably hacky.

As stated above you can access the data by fiddling with the PC, but this is a nasty workround (IMO).

Clearly it has not been a priority for the maintainers.