micropython/micropython #1143: Inline assembler: conditional branch fails in long programs

peterhinch, 2015-03-02:
This issue has been present for some time but the improvements to the error reporting now make it possible to figure out what's going on. Unfortunately it can only be replicated in a long program: see [here](https://github.com/peterhinch/micropython-fft.git).

The beq() at line 401 produces a "branch not in range" error even though the target is local. If beq() is replaced with b() to the same target the code imports and runs correctly.

dpgeorge, 2015-03-02:
bcc ops (such as beq) have a 9-bit range (ie 512 signed words, ie can reach around 256 instructions before or after it).  You need to use the wide version of the op for long jumps, which is now implemented: beq_w(label).

closed 2015-03-02 by dpgeorge

peterhinch, 2015-03-02:
Perhaps I'm missing something here but I don't think this explains what I'm seeing. The beq() at line 401 only skips 24 instructions (the label is on line 429). Why am I seeing the  "branch not in range" error here? And why does beq(DFTDONE) fail when b(DFTDONE) does not? Both have the same 9-bit range so surely they should be subject to the same limit.

dpgeorge, 2015-03-02:
The "b" instruction has a 12 bit range, which might help explain things.  But you're right that DFTDONE should be in 9-bit range...

dpgeorge, 2015-03-02:
Ok, found the bug and fixed in 67c5f89af539e0778155dcd35b58c32af64debec: it was giving an error on the first pass of the assembler, when it only knows the correct branch value on the last pass.

peterhinch, 2015-03-03:
Great work! I seem unable to break it now :)