micropython/micropython #1146: [Question] have you ever seen the Synapse SNAP?

cloudformdesign, 2015-03-04:
http://www.synapse-wireless.com/snap-components-free-developers-IDE-tools/portal

They are an embedded microcontroller with python, similar to micro python. Of course, they aren't open source and they use a proprietary wireless network -- but they are still cool. They are your precursors!

If I had one wish it would be to get micropython on an integrated ZigBee chip to implement all the mesh networking + low power of SNAP on a standard and open wirless protocol. If I had a second wish, it would be to implement ZeroMQ on both wifi and Zigbee nodes for simple and robust pub/sub communication.

pfalcon, 2015-03-05:
Yes, sure, we heard about SNAP. I wouldn't call it MicroPython's precursor though, such would rather be PyMite. But yes, that shows the evolution: tools which used to be proprietary and niche are now open and available for reuse for many different purposes in different areas.

cloudformdesign, 2015-03-06:
Oh ya,  I forgot about pymite.

I've been thinking recently,  can you point me to the code you use for your memory manager? I'm thinking of developing my own,  and I want to see your implementation

dpgeorge, 2015-03-09:
> can you point me to the code you use for your memory manager?

All code can be found in this repository.  Memory management can be customised per port, but most ports use the built-in garbage collector, found in py/gc.c.

ZeroMQ would be nice to support, and feel free to have a go at implementing it!

closed 2015-03-09 by dpgeorge

cloudformdesign, 2015-03-11:
@dpgeorge I think I am starting to understand the garbage collector, but it would be helpful if there was a document giving an overview of the structure.

From looking at the code, I would glean the following:

## User Level Functions
The main functions that a user needs are:
`gc_alloc` -- malloc with gc
`gc_free` -- free with gc

These are built on top of a memory structure using Allocation Table Bytes (ATBs)
## Memory Structure
Memory is split up into 4 Allocation Tables. Every Allocation Table has a `ATB` which stands for "Allocation Table Byte". This is a single byte containing 4 sets of the following
```
// 0b00 = FREE -- free block
// 0b01 = HEAD -- head of a chain of blocks
// 0b10 = TAIL -- in the tail of a chain of blocks
// 0b11 = MARK -- marked head block
```
These are known as `ATB_0` through `ATB_3` and have several C methods (i.e. functions and macros) to access their attributes. These include:

BLOCKS_PER_ATB  -- The number of ATB's that fit in an Allocation Table
ATB_MASK_0 -- Get the relevant bytes for ATB_0
...
ATB_MASK_3 

ATB_0_IS_FREE(a) -- Determine whether table is currently free
...
ATB_3_IS_FREE(a)

** What do  these do? Why are they useful??? **
BLOCK_SHIFT(block)
ATB_GET_KIND(block)
ATB_ANY_TO_FREE(block)
ATB_FREE_TO_HEAD(block)
ATB_FREE_TO_TAIL(block)
ATB_HEAD_TO_MARK(block)
ATB_MARK_TO_HEAD(block)

BLOCK_FROM_PTR(ptr) 
PTR_FROM_BLOCK(block) 
ATB_FROM_BLOCK(bl)

I have to go for now, I will update this post with my actual questions soon

dpgeorge, 2015-03-11:
> I think I am starting to understand the garbage collector, but it would be helpful if there was a document giving an overview of the structure.

Sorry but there is no time/resources for such things.

It's a standard mark-sweep garbage collector with optimisations for memory constrained systems.

> I'm trying to implement a library called tinymem that will address most of these concerns

Are you writing a garbage collected system, or something else?

cloudformdesign, 2015-03-11:
Yes, a fast low memory garbage collected system, ideally with defragmentation support. This means that the user would actually hold a "pointer to a pointer" that would never change, and the actual location of their data could change. This allows the underlying memory to be defragmented to free up large chunks of memory if it has become too fragmented to get large amounts of data.

It will also have special support for lower chunks of memory ( < 32 bytes) to have very little memory overhead for obtaining those pointers.

There are obviously several issues with this, all of which I am trying to address -- the main one being that the user has to deal with type casting `void` pointers.

pfalcon, 2015-03-11:
@cloudformdesign: There's a forum at http://forum.micropython.org/ which may be more suitable for asking questions/discussing things than the bugtracker here. It will give better exposure and if you search it first, you may find existing discussions on related topics which may give some insight, or at least chance to cooperate with other users interested in that area.

Otherwise, I can second what @dpgeorge : gc.c is almost a textbook implementation of mark-sweep GC, and you shouldn't have much trouble understanding it (after spending some time on it of course) if you're after developing much more sophisticated memory management.

cloudformdesign, 2015-03-11:
In your opinion can the gc architecture be used for native C types? I.e. could I use it to allocate memory for a C extension? I'm guessing that I couldn't

cloudformdesign, 2015-03-11:
thanks @pfalcon will do!

dpgeorge, 2015-03-11:
> In your opinion can the gc architecture be used for native C types?

Yes, it's used all the time for this.  The GC is just an implementation of malloc/realloc/free.