micropython/micropython #1147: STMF4DISC ceased working after 534574348e477005b8647e45402d859a0f3db6ca commit

siorpaes, 2015-03-06:
Hello,

I updated to latest available commit yesterday but unfortunately STM32F4-Discovery port stopped working: no CDC nor Mass Storage Device appeared over USB. I double checked with 'lsusb' with Linux but no VID/CID pop up. If I reset the board with no USB cable inserted, the green LED (LD4) switches on and after 7 seconds all other LEDs start blinking twice in circles and finally end up in being all on. If I reset the board with the USB cable connected, green LED stays on and all other LEDs stay off.

So, I gradually rolled back to previous versions and sorted out that this effect is happening from 534574348e477005b8647e45402d859a0f3db6ca version onward. Differences with previous version 11aa6ba456287d6c80598a7ebbebd2887ce8f5a2 are not giving me any hint to me as they appear to have nothing specific to STM32F4DISC port.
Do you have any suggestion on where the issue could be ?

For a rapid test, firmware images associated to the two hashes are available here:
https://www.dropbox.com/s/ue9txaz36ou5w0q/STM32F4-Discovery.zip?dl=0

Many thanks in advance,

David

dpgeorge, 2015-03-06:
Do you have any boot.py or main.py files in the local FS?

siorpaes, 2015-03-06:
Hello,

I wipe all MCU flash, then upload the DFU image using DfuSe tool from ST (also tried uploading the .hex file with ST-Link debugger with same results).

In the working 11aa6ba commit I have both boot.py and main.py.

main.py is empty, whereas boot.py contains:

```
# boot.py -- run on boot-up
# can run arbitrary Python, but best to keep it minimal

import pyb
#pyb.main('main.py') # main script to run after this one
#pyb.usb_mode('CDC+MSC') # act as a serial and a storage device
#pyb.usb_mode('CDC+HID') # act as a serial device and a mouse
```

In 5345743 version I can't access to the File System.

dpgeorge, 2015-03-06:
Try compiling with the following changes to stmhal/mpconfigport.h:
```diff
-#define MICROPY_EMIT_THUMB          (1)
-#define MICROPY_EMIT_INLINE_THUMB   (1)
+#define MICROPY_EMIT_THUMB          (0)
+#define MICROPY_EMIT_INLINE_THUMB   (0)
```

siorpaes, 2015-03-06:
Yes, with these changes it works now

dpgeorge, 2015-03-06:
Then you must have something/some code that uses either @micropython.native or @micropython.viper or @micropython.asm_thumb.  Are you sure your boot.py and main.py are empty?  Have you otherwise modified any of the source code?

siorpaes, 2015-03-06:
Yes, main.py and boot.py are empty. I am doing a clean clone from the Git repository and checking out the versions with 

```
git checkout <SHA>
```

siorpaes, 2015-03-06:
Just checked with another STM32F4-Discovery board (as well as using other OS for DFU, though): the very same DFU binary 5345 uploaded to Dropbox before works like a charm on it!
Going to investigate further on what's going on with the offending board (or OS) and keep posted if I have any interesting news.
Pls close the issue, sorry for the bother and thanks for your prompt support ;)

closed 2015-03-07 by siorpaes

siorpaes, 2015-03-09:
Some news: the very same binary image (full flash 1MB dump) works correctly on one board but crashes on another one. Tried to enable debugging facility so to nail down the problem but... it started to work again (argh). Adding some other debugging code in __fatal_error() also ended up in this Heisenberg effect, so I start doubting that EMIT_THUMB is the direct cause of the issue..
After some more debugging discovered at least that it hits the HardFault handler.

dpgeorge, 2015-03-09:
Does going from a working to non-working image make the firmware size cross some interesting value, such as 256k?

siorpaes, 2015-03-09:
Yep, suspected this and did a check but seems not: working firmware1.bin is 247716 bytes, non-working firmware1.bin is 247832 bytes. Though, a 514172 bytes image (with debugging enabled) is not crashing.
By snooping at the CFSR register it looks like the HardFault is generated by an Undefined Instruction. Hope I am able to recover the offending PC from the stack.

siorpaes, 2015-03-10:
Some more news: the Undefined Instruction exception happens randomly between 0x0803c520 and 0x0803c52a which correspond to memset+50 and memset+60. Occasionally, in gc_init+54 (right after memset).
Must definitely have a broken hardware...

dpgeorge, 2015-03-11:
It could be that the addition of EMIT_THUMB code pushed critical/used code (memset) into a region of flash that is bad and now you are seeing that effect.  Did you try getting the flash contents and verifying they are the same as the image you downloaded?

siorpaes, 2015-03-11:
Checked this with md5s: flash looks working fine. So, I tried to stimulate that very memset sitting in that same position by calling it from a test firmware I built on purpose with lengths ranging from 0 to 124k and.. it works like a charm (!). I think I ran out of ideas :(

pfalcon, 2015-03-11:
@siorpaes: If I were in your position, I would suspect compiler optimizations. (So would tell what compiler I use, posted gcc -v output from it, tried to compile with different -O flags, etc.)

GuyCarver, 2015-03-11:
Perhaps you ran out of Stack space?

siorpaes, 2015-03-11:
Hi @pfalcon and @GuyCarver, many thanks for your inputs.
I am not sure how a stack overflow or compiler optimization option or any other software-related variable could determine different firmware behaviors on two identical hardware implementations. The very same firmware binary image is working perfectly on one STM32F4-Discovery board but it's crashing on another identical one. If there was any software issue, I should see both boards crashing. If there isn't any software issue (as it seems, as I'm the only one experiencing this crash so far), I should see the firmware running merrily on both boards, isn't it ?

dpgeorge, 2015-03-11:
Latest master version runs fine on my Discovery board.

siorpaes, 2015-03-12:
As the flash looks behaving correctly, I am suspecting a flash prefetch malfunction: disabling this feature solves the issue. Here's an interesting report of some guys experiencing similar behavior http://goo.gl/r9s1au

siorpaes, 2015-03-12:
Problem vanishes also increasing flash Wait States as suggested in above link (FLASH_LATENCY_6 is enough).
Guess I can end here my debugging sessions :)

dpgeorge, 2015-03-12:
Very interesting.  So it's a silicon issue, or is the default wait state too low?

siorpaes, 2015-03-12:
The official clock tree configuration tool CubeMX from ST sets the wait state configuration to 5 for SYSCLK=168MHz, so I suspect silicon issue. I will also try with lower system clock settings and cooling down the chip and see what happens.

siorpaes, 2015-03-12:
Had some fun in cooling the chip with an ice cube and can confirm that temperature does matter! I'll publish a video on YouTube ;)

siorpaes, 2015-03-14:
Even better using a freezing spray to cool down MCUs... here you can see how temperature does matter: http://youtu.be/ZouqXRgO3ow

dpgeorge, 2015-03-16:
Cool!

I noticed that uPy takes ages to boot on the F4 discovery, and that's because it's waiting for the LSE to start up (which will never happen since there is no external crystal for the RTC).  In 49fe6dc89a45a5791cfa8dad86f25469749f2312 I have fixed this by adding a config option to use the LSI instead.

siorpaes, 2015-03-17:
Wow, it's booting super-fast now :) thanks for this update.