micropython/micropython #1149: ure library - put in rubbish make it crash

clacktronics, 2015-03-06:
I accidentally put in a bad regular expression and made it crash. I paired it down to this.

```python
import ure
rubbish = ure.match('+','test')
```
I get a 4 LED fail.
Build is v1.3.10-147-g16b1f5e

let me know if I am being annoying, I am not sure if I am following the etiquette

blmorris, 2015-03-06:
Re: being annoying - writing as another user who posts more problems than solutions, my experience is that the developers do prefer to have bug reports posted as issues here - and a hard crash like this certainly counts. I think the proper etiquette is being observed ;)

clacktronics, 2015-03-06:
Thanks!

I have found something else but is not a hard crash. An escaped dash "\-" does not return. I tried it out in regular python to check and it works as expected there ( there are massive differences between re and ure though )

```python
Micro Python v1.3.10-147-g16b1f5e on 2015-03-05; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import ure
>>> ure.match('[a-zA-Z\-]*','test-one').group(0)
''
>>>
```

pfalcon, 2015-03-06:
> let me know if I am being annoying, I am not sure if I am following the etiquette

ure is based on the regexp code which was intended to show how to implement regexp matching, not to be production ready, so it misses many boundary conditions checks. And reporting issues people hit while using ure in real life is the only way to fix them. So yes, any bugreports, accompanied by info needed to reproduce an issue, are much appreciated and will be acted upon (sooner or later).

pfalcon, 2015-03-06:
> there are massive differences between re and ure though

Indeed there're, and ure doesn't try to implement complete re functionality, but rather some minimal subset, based on generic definition of regexp. And with that generic definition, to put hyphen literally in character set, it should be first, e.g. `[-A-Z]`. In any other position, it can be interpreted as character range designator by an implementation.

pfalcon, 2015-03-06:
(And please report different issues in different tickets.)

dpgeorge, 2015-03-10:
Crashing on random input should now be fixed by 8dead2a6c64e8589e06e1e218d40e63aea18a009.

closed 2015-03-10 by dpgeorge