micropython/micropython #1160: Confusing comment regarding map lookup types

pfalcon, 2015-03-16:
https://github.com/micropython/micropython/blob/master/py/obj.h#L159 says: 
~~~~
// These can be or'd together
typedef enum _mp_map_lookup_kind_t {
MP_MAP_LOOKUP, // 0
MP_MAP_LOOKUP_ADD_IF_NOT_FOUND, // 1
MP_MAP_LOOKUP_REMOVE_IF_FOUND, // 2
} mp_map_lookup_kind_t;
~~~~

But lookup types are neither actually OR'ed together, nor it makes sense to. @dpgeorge: Do you agree? And in some places, they're still tested using &, that should be cleaned up too then.

dpgeorge, 2015-03-16:
set_symmetric_difference_update seems to be the only guy that uses the or'ing feature.  So probably make that comment refer to set operations only, and clean up the map function to not use &.

pfalcon, 2015-03-16:
I see, thanks for the reference! But then I suggest to make the comment more clear and verbose, like move "// These can be or'd together" to be above MP_MAP_LOOKUP_ADD_IF_NOT_FOUND, hinting that it's MP_MAP_LOOKUP_ADD_IF_NOT_FOUND and MP_MAP_LOOKUP_REMOVE_IF_FOUND can be or'ed, and make the comment more explicit, like "// These can be or'd together, meaning that item will be added if not exist, and removed if exists".

Otherwise, simple logic says that you can't add and remove at the same time, so or'ing opportunity seems confusing.

dpgeorge, 2015-03-16:
Better would be to have a predefined constant `MP_MAP_LOOKUP_ADD_IF_NOT_FOUND_REMOVE_IF_FOUND=3`.  Then it needs no comment (except that you can't pass this to the map function, only works for set).

closed 2015-03-20 by dpgeorge