micropython/micropython #1165: stmhal: I2S class for pyb module (RFC)

blmorris, 2015-03-19:
I am beginning to work on a new class to support I2S (Inter-IC-Sound) in uPy. I am admittedly in over my head but now making some steady progress - using spi.c as a model I have managed to create a skeleton I2S class that - seems to - successfully initialize the SPI2 / I2S2 peripheral in I2S mode with a hard-coded list of pins corresponding to those used on my DSP-amplifier board. It does not yet initialize any clock parameters or implement any data transfer functions.
I would like to get some early feedback because the usage pattern (and therefore the API) for I2S is going to be significantly different from SPI even though they are based on the same peripheral hardware in the STM32F4xx processors, and I would like to avoid implementing anything that is too ugly or awkward to merge into the project.

To illustrate the difference between I2S and SPI usage, here are some tasks that I expect to implement with I2S (all of which we can currently do at some level with firmware developed in C, but uPy will give us more flexibility to develop advanced applications.)
* Read audio data (.wav file or other format) from SD card and send data stream via I2S to audio codec chip, (possibly implementing sample-rate conversion on the way.) 
* Record audio data from same codec chip to SD
* Receive data from codec, apply filtering, and concurrently send back out to codec for playback.
* Send test signal to codec for playback and concurrently receive data from a microphone via the codec for room acoustic analysis.

All of these cases imply that the I2S data transfer gets set up and runs indefinitely, unlike the data bursts usually seen in SPI - DMA is probably even more critical here than for SPI. Also, it is important to note that only instances 2 and 3 of the SPI hardware support I2S, so it doesn't neatly map to the X,Y position scheme of the pyboard. On the pyboard, as well as my own board, SPI/I2S3 are all but inaccessible due to conflict with the SD card and LED's. On the other hand, the only common platform for most others to easily use I2S with uPy is the STM32FDISCOVERY board, which has a codec available on I2S3.

My initial proposal for the API would be to initialize I2S with a list of pins rather than an instance or position number (necessary for me because the pins used for I2S on my own board are not the same as the SPI2 pins on the pyboard.) The list would have the form `[clock, word_select, tx_pin, rx_pin]`. The init function will check that each pin is valid for the function designated by its position, it will check that all pins are members of the same I2S instance (2 or 3) and check that the I2S instance is available and not already in use by an existing SPI object (this last check may not be strictly necessary). The only other required parameter will be whether it is a master or slave, and the clock configuration parameters if it is a master. All other parameters will be defined by kwargs set to some reasonable defaults - probably the ones used by my own project ;-)
The list of pins will implicitly determine whether I2S operates in simplex or duplex mode (if either rx_pin or tx_pin == 0 then that direction is omitted.) Likewise, the position of the pin name in the list determines which of the data pins is an input and which is an output.
I'm still trying to decide whether to pass in a list of pin objects or simply pin descriptor strings (i.e. wether to pass `pyb.Pin.cpu.B10` or just `'B10'`)  
 
I'm going to have more questions along the way so I would hope to keep this ticket open for discussion.
-Bryan

dpgeorge, 2015-03-20:
I don't know much about I2S, but from looking at the datasheet it seems you can have an extra pin, MCK (master clock).  Is this used much?

I agree with everything else you say above.  For pin selection, you can support both pin objects and strings if you pass the argument through the `pin_find` function.  It'll return a proper pin object.

blmorris, 2015-03-21:
You are right, the MCK and I2S_CKIN pins should both be supported as available options. MCK is an optional output that usually has a frequency of 4-8x the bit clock (BCK or just CK), it can provide the primary clock input for another chip, eliminating the need for that chip to have a dedicated crystal. Likewise, I2S_CKIN allows the dedicated I2S PLL on the STM32F4xx to be driven directly by an external clock source, usually same the clock that drives any external audio ADCs/DACs/codecs.
According to my notes (I should double-check), for the pyboard's cpu MCK is available on only one pin for each I2S interface (PC6 for I2S2 and PC7 for I2S3). The only pin that can be configured as I2S_CKIN is PC9, which is used by the SD card interface and is therefore unavailable on the pyboard as well as on my board. All the same, it ought to be supported for completeness and portability to other projects.
Thanks for the tip about `pin_find` - I was thinking about the logic for accepting pin arguments, and I hadn't found that yet.

blmorris, 2015-04-08:
I have posted my current I2S work here: https://github.com/blmorris/micropython/tree/stm-i2s
It isn't anywhere close to being ready to merge, but I can make it into a pull request if that makes it easier to review. (Also I'm not sure what is up with the commit dates - probably a result of squashing all my intermediate commits together).
At this point my code provides an I2S object that can configure most of the parameters of the I2S hardware (though I can't test most of these configurations at this point) and it will activate the hardware in master or slave mode. The send and receive methods are currently just stubs; I plan to work on those as well as implementing DMA next.
The API as currently implemented may be overly complex, but there are many options with I2S and it is hard to justify arbitrarily cutting them out of the API. The options are described in detail in the comments of i2s.c, but here is a quick synopsis showing how an I2S object can be created with a (valid) list of pins, and how the object print function works.
```
Micro Python v1.4.1-20-g32fb27f on 2015-04-08; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> i2s = pyb.I2S(['B10','B12','C2','C3'])
>>> i2s
I2S(2 on [B10 B12 C2 C3])
>>> i2s.init(i2s.MASTER)
>>> i2s
I2S(2 on [B10 B12 C2 C3], I2S.MASTER, MCLK off, freq=48000, standard=0, format=1, polarity=0)
>>> i2s.init(i2s.MASTER, mclkout=i2s.ENABLE)
>>> i2s
I2S(2 on [B10 B12 C2 C3], I2S.MASTER, MCLK on C6, freq=48000, standard=0, format=1, polarity=0)
>>> i2s.init(i2s.SLAVE)
>>> i2s
I2S(2 on [B10 B12 C2 C3], I2S.SLAVE, standard=0, format=1, polarity=0)
>>> 
```
I would welcome any tips, comments or critiques - software development is not my primary skill, and I'm pretty sure that I never learned a lot of this stuff in my introductory C class many years ago ;)
Thanks!

blmorris, 2015-04-10:
I have tracked down something that looks suspiciously a hardware bug in the I2S port on my pyboard; I would gratefully appreciate if anyone else could help with a sanity check on my code. I should say that I haven't found any other reference to a similar problem on the ST forums.
In the MicroPython branch that I posted above, you can run the following code:
```
i2s = pyb.I2S(['B10','B12','','C3'])
i2s.init(i2s.MASTER)
```
This code will select and configure I2S2 / SPI2 (because the pins are all associated with I2S2; I2S3 pins are inaccessible on the pyboard so it is disabled by macro). B10/Y9 gets configured as the Bit Clock, B12/Y5 gets configured as the Word Select clock, and C2 / X21 gets configured as SD receive. Based on the default values that I have set, the Bit Clock will run at 3.072 MHz and WS will run at 48 kHz. All well and as intended; I can confirm that the following code gets run and sets the mode parameter correctly for HAL_I2S_Init().
```
    // init->Mode can be MASTER_TX, MASTER_RX, SLAVE_TX, or SLAVE_RX;
    if (args[0].u_int == 0) {
	init->Mode = self->base_is_tx ? I2S_MODE_MASTER_TX : I2S_MODE_MASTER_RX;
    } else {
	init->Mode = self->base_is_tx ? I2S_MODE_SLAVE_TX : I2S_MODE_SLAVE_RX;
    }
```
The following code also sets the mode parameter correctly, in this case with the Data pin in Transmit mode: (Two data pins can be passed to the object constructor, the first is transmit and the second is receive.)
```
i2s = pyb.I2S(['B10','B12','','C3'])
i2s.init(i2s.MASTER)
```
In this case, however, the clock signals disappear. The puzzling thing is that I can confirm that the I2S configuration register gets the intended value! Once the I2S code is running, I can use `stm` to directly read and write the SPI_I2SCFGR register:
```
# This will set RX Master mode, and the clocks will be on:
>>> stm.mem32[stm.SPI2 + stm.SPI_I2SCFGR] = 0xf01
# This will set TX Master mode, but the clock signals will be
# turned off even though they should be on:
>>> stm.mem32[stm.SPI2 + stm.SPI_I2SCFGR] = 0xe01
```
My reference is the programmer's reference manual, RM0090 (version 7) section 28.5.8 - SPI_I2S configuration register (SPI_I2SCFGR)

Thanks!

dpgeorge, 2015-04-11:
By "the clock signals disappear" do you mean that B10 no longer outputs a clock?

Did you try (after a fresh reset) creating the second version of I2S first (with config value 0xe01)?

Did you notice in the datasheet that the I2S peripheral must be disabled before configuring it?  The HAL seems to do that for you so I doubt that's the issue...

blmorris, 2015-04-11:
> By "the clock signals disappear" do you mean that B10 no longer outputs a clock?
Yes, the clock signals stop on both B10 and B12.
> Did you try (after a fresh reset) creating the second version of I2S first (with config value 0xe01)?
Yes, that is how I initially discovered the issue. It turns out that I can effectively turn the clock outputs on and off by switching the I2S configuration and re-initializing in the two different modes - as far as I can tell, the clock outputs should be present in both Master modes (Master Tx and Master Rx) and be off in Slave Tx and Slave Rx modes.
> Did you notice in the datasheet that the I2S peripheral must be disabled before configuring it? The HAL seems to do that for you so I doubt that's the issue…
Yes, I considered that. As you said, the HAL does this so it doesn't seem to be the issue. However, I can also turn the clock outputs on and off simply be toggling the I2SCFGR register between 0xf01 and 0xe01, without disabling the I2S.

I'm starting to suspect that I must be missing something in the configuration, and that the weird thing is that the clock is working int Master Rx rather than it being missing in Master TX. 
Simplex Master Tx should be the most basic mode to configure and use, so there would be lots of reports online if the hardware was that broken. My next step will be to see if I can test a very simple I2S initializer without the uP layers, and see if I can find the bug that way.

Thanks!

blmorris, 2015-04-14:
I tracked down the problem with I2S Master Tx mode. Turns out that my code was correct and the hardware was working exactly as it should have done all along - the only problem was a multiple head-desk worthy misinterpretation on my part.
Explanation: when an I2S Master is in Receive mode, it need to provide the bit clock and the frame (word-select) clock at all times so that the slave device can push data on to the bus. There is no handshaking or error checking; if the slave just pushes out zeroes (or if it isn't even there!) the constant low level on the data line is still perfectly valid data: it is just valid data consisting entirely of zeroes.
Master Transmit mode is different - if the processor isn't actively transmitting data, then it doesn't need to provide clock signals to drive the slave. If the goal is to send a continuous stream of zeroes in Master Tx mode, then a steady stream of zeroes needs to be send to the transmit register; otherwise, the clock signals will stop.
That is why Master Rx mode produced continuous clock signals and Master Tx mode turned them off.  Now that I now everything is working as intended, I can flesh out the full send and receive methods and start working on DMA.

blmorris, 2015-04-14:
Having resolved my issue with Tx Master mode, I'm now trying to implement the I2S send and receive methods, hopefully with DMA, and I have a few new questions.  I'm basing my work on stmhal/spi.c; according to the programmer's reference DMA works the same with I2S as with SPI, so I hope that the adjustments will be fairly small. I'm not there yet, but I wonder if it will make sense to eventually split the DMA code into separate shared files, spi-i2s-dma.[ch]
One difference between I2S and SPI (in the current uPy implementation) is that SPI assumes that data is 8 bits wide; I2S will need to get data into the transmit buffer in 16 bit chunks. (24 and 32 bit samples are also possible, but they are broken up into two 16-bit chunks for each sample transmitted or received.) I can't tell what I need to change in the bufhelper routines to allow this.
Also, in my own pyb_i2s_make_new function, the current implementation accepts a list of pin descriptors as it's required argument (these can be name strings or full pin object names.) I would like to add an option so that if pyb.I2S() is called without any arguments, it defaults to creating a new full-duplex I2S port using the default pins for SPI2 (Y5, Y6, Y7, Y8), to do this I need to figure out how to create a python list in C that I can pass to the pin validation logic.
Thanks!

blmorris, 2015-04-14:
Regarding my last post- it appears that the 8-bit limitation is not in the bufhelper routines but rather in the DMA config that I copied over from SPI. When the DMA stuff is removed I can transmit 16-bit data.

dpgeorge, 2015-04-14:
> Regarding my last post- it appears that the 8-bit limitation is not in the bufhelper routines but rather in the DMA config that I copied over from SPI.

Yes, bufhelper can get arbitrary types of data.  It's just data.  I'm sure the MDA can be configured to work in 16 bit mode.

> I need to figure out how to create a python list in C

Schematically it would be something like:
```c
mp_obj_t tuple[2] = {&pin_Y1, &pin_Y2};
mp_obj_t o = mp_obj_new_tuple(2, tuple);
```

blmorris, 2015-04-15:
Thanks Damien, that was the cue I needed. I got this to work, now the constructor called with no arguments returns an I2S object on the pyboard's SPI2 port:
```C
// get array of pin identifiers, could be name strings or pin objects
mp_buffer_info_t bufinfo;
mp_obj_t *pin_names;
const pin_obj_t *pins[4];
if (n_args == 0) {
#ifdef PYBV10	
    const pin_obj_t *dflt_pins[4] = {&pin_B13, &pin_B12, &pin_B15, &pin_B14};
    mp_obj_t dflt_pin_tpl = mp_obj_new_tuple(4, (mp_obj_t)dflt_pins);
    mp_obj_get_array(dflt_pin_tpl, &bufinfo.len, &pin_names);
#endif
} else {
    mp_obj_get_array(args[0], &bufinfo.len, &pin_names);
}
```
I pushed the update to my stm-i2s branch.

blmorris, 2015-05-21:
I have made some progress implementing DMA support in my I2S class; there is still much to do, but I have just had success transmitting a buffer in DMA mode and returning to the REPL.
I was stuck for a while where I could transmit data but the board would freeze up and not return to REPL without a hard reset - it turned out that I had nearly everything in DMA configured correctly (essentially copied over from `spi.c` with a few I2S specific changes).
However, when I initially tried to compile, I encountered this error:
```
LINK build-PYBV10/firmware.elf
build-PYBV10/i2s.o: In function `DMA1_Stream3_IRQHandler':
i2s.c:(.text.DMA1_Stream3_IRQHandler+0x0): multiple definition of `DMA1_Stream3_IRQHandler'
build-PYBV10/spi.o:spi.c:(.text.DMA1_Stream3_IRQHandler+0x0): first defined here
build-PYBV10/i2s.o: In function `DMA1_Stream4_IRQHandler':
i2s.c:(.text.DMA1_Stream4_IRQHandler+0x0): multiple definition of `DMA1_Stream4_IRQHandler'
build-PYBV10/spi.o:spi.c:(.text.DMA1_Stream4_IRQHandler+0x0): first defined here
make: *** [build-PYBV10/firmware.elf] Error 1
```
Okay, easy enough, just comment out these two lines in `i2s.c`:
```C
#define I2S2_RX_DMA_IRQ_HANDLER DMA1_Stream3_IRQHandler
#define I2S2_TX_DMA_IRQ_HANDLER DMA1_Stream4_IRQHandler
```
Great, now it compiles and seems to work - only it locks up after transmitting the buffer and doesn't return to the REPL - of course it doesn't, because the I2S DMA IRQ handler doesn't have the correct hook, and the default handler just spins in an infinite loop.
(It took me a few days and a lot of help to figure that out…)
My solution for now is to comment out the same lines in `spi.c` so that I can work with I2S, but obviously that just breaks SPI. Given that there is so much overlap between SPI and I2S (they use the same serial hardware, just configured differently) the best way to resolve this is to refactor the common functions into a new file with shared names, but I'm not sure how best to proceed with that.
I will post my updated I2S code after a bit of cleanup.

dhylands, 2015-05-21:
We should probably create a dma handler, and spi and i2s would register with the dma handler (and pass in a function pointer) The actual irq handlers would exist in the dma handler.

fojie, 2015-05-21:
@blmorris: Hi, I found a problem with SPI becoming very slow under some circumstances, see issue #1268. My current hypothesis is that it might have something to do with buffer,  DMA / IRQ because it seems related to whether using Network version (cc3k), nr bytes sent and chosen baud rate. Since you're looking at DMA and IRQ, you might stumble across something in the code (that has recently changed between network versions 1.4.2 and 1.4.3) that explains my findings.
I'm for the moment trying to isolate some small code example that clearly shows the problem.

dpgeorge, 2022-03-30:
I2S support for stm32 was added in 8a5bfe44a5

closed 2022-03-30 by dpgeorge