micropython/micropython #1168: Make garbage collector built on malloc / free implementation with pointer indirection

cloudformdesign, 2015-03-26:
This is a point that was brought up in the [tinymem forum thread](http://forum.micropython.org/viewtopic.php?f=3&t=591), where @pfalcon said:

> Making uPy GC pluggable, so it can work with system heap (or any external memory manager) is a worthy aim to follow.

This issue is meant to address this, while simultaneously addressing a separate issue associated with the tinymem library -- that it uses pointer indirection to do it's defragmentation, and therefore all calls to pointers need to be wrapped in a `tm_void()` function call (in some way)

The way to solve this would be:
- make the underlying structure of the memory manager use a macro like `base_alloc` and `base_free` that could be adjusted depending on the memory manager used
- have all pointers pass through macros. So, for example, instead of saying `uint8_t *dynamically_allocated_pointer` you would say `memptr_uint8_p(index)` -- which would convert it to a pointer for you.

pfalcon, 2015-03-26:
See https://github.com/micropython/micropython/pull/1166 & https://github.com/micropython/micropython/issues/1161 .

cloudformdesign, 2015-03-26:
oh, so does #1166 already do this (it sure seems to)? I did a search for it but didn't see that issue.

Also, I think #1161 is more related to #1169, but that issue is more specific (although would accomplish the same goal)

pfalcon, 2015-04-05:
> oh, so does #1166 already do this (it sure seems to)? I did a search for it but didn't see that issue.

Just to clarify, #1166 doesn't "already do this", it lays out framework of being able to do that, and make few steps in implementing it. Dozens, possibly hundreds steps are needed to have it really "done".

cloudformdesign, 2015-04-05:
Got it.  I hope by next weekend to be able to start making pull requests on micropython to accomplish parts of these goals. I'm going to start with this one.

pfalcon, 2015-04-05:
Sounds good. I suggest you start with "Make garbage collector built on malloc / free" part. That's actually doable in one setting and if code is clean, mergeable. Anything else is terra incognita and will have too spend a lot of time in work, rework, review, rinse and repeat dozen times.

cloudformdesign, 2015-04-14:
@pfalcon I'm taking a look at the `gc.c` source code right now, and it looks like our main goals are the following:

It looks like there are some major changes that have to be made in order to implement traditional malloc/free archetecture:
- pointers to blocks are stored in a table, not a pool of all blocks (malloc/free is managing the pool)
- change anything that relies on `MP_STATE_MEM` to instead rely on a malloc/free
    - for instance, `gc_alloc` begins it's process by finding the number of "blocks" that are needed and then looks for a "run" of that many available blocks. I'm pretty sure this whole function is just a basic `malloc` implementation with memory-reuse.

If that is what is necessary, it is going to have significant memory ramifications right off the bat -- I'm hoping you have another perspective/solution?

Btw, I've just made a major push to tinymem, including some pretty serious unit tests that stress it pretty extensively. Check it out:
https://github.com/cloudformdesign/tinymem

cloudformdesign, 2015-04-14:
Also, @pfalcon, @dpgeorge  I would love to do a google hangouts sometime if you had 15 minutes. I would like to just give a quick overview of what tinymem does and make sure it meets your requirements.

dpgeorge, 2015-04-14:
@cloudformdesign sounds good, but I predict it'll be difficult to find a time when we are all free...

cloudformdesign, 2015-04-14:
Oh, sorry I didn't mean we **had** to do it all together, although that would be great. 

Also, I'm pretty available as I just got laid off, haha.

pfalcon, 2015-04-15:
> there are some major changes that have to be made

Yes, some refactoring is required to make uPy's GC interface more clean and implementation pluggable. (And that should be the same - to allow to plug in any GC implementation, not just malloc/free).

> change anything that relies on MP_STATE_MEM

MP_STATE_MEM is internal structure of GC algo. Or rather, it consists of 2 parts: one for stats which is generic, another for internal GC algo data, which will be own for each GC algo. Complete refactor thus would need to make this split.

The rest are implementation details of a particular GC algo, and they are different for each algo, and don't need to correlate. So, I don't see why it's going "to have significant memory ramifications right off the bat". To remind, to implement GC, you need: way to enumerate all allocations, way to identify pointers within each allocation, way to mark seen objects, initial set of allocations to start from. For "blackbox malloc" case, that reduces to having 2 word overhead per allocation (to store linked list pointer and block length, mark bit can live in a pointer which is guaranteed to be aligned to power of 2 bigger than 1 for any realistic malloc).

pfalcon, 2015-04-15:
> google hangouts sometime if you had 15 minutes. I would like to just give a quick overview of what tinymem does

I'd suggest to make a blog post about that, and contrast your implementation with existing algorithms. What's wrong with existing algos, what needs be improved, how that can be improved and why do you think it'll be improvement instead of not. That way, not just 2 people in the world can get to know that, but much wider audience.

> and make sure it meets your requirements

When you say "your" do you mean "your personal" or "requirements of micropython"? Re: 1st case, I don't think that any of us has some very specific requirements to discuss without seeing a working implementation. Re: 2nd, as you can see, uPy doesn't yet have any specific interface requirements either - it simply has one memalloc+GC algo hard-coded. That needs to be changed first.

cloudformdesign, 2015-04-15:
@pfalcon 
>  I don't think that any of us has some very specific requirements to discuss without seeing a working implementation

Well that's what I want to discuss! I have a working early implementation!

cloudformdesign, 2015-04-19:
@dpgeorge It sounds like pfalcon won't be doing it anyway. Are you still interested? If you are, send me an email at googberg@gmail.com

cloudformdesign, 2015-04-26:
I have obviously been searching for a long time for a decent memory manager for microcontrollers, and I think I have finally found one that I would consider decent:

https://github.com/dimonomid/umm_malloc/blob/master/umm_malloc.c

This stackoverflow article details it:

http://stackoverflow.com/questions/177209/whats-a-good-c-memory-allocator-for-embedded-systems/29882348#29882348

Since this memory manager uses a traditional malloc/free approach (does not require pointer indirection) and is extremely well documented, I think it would be a good place to start. 

Note that it does NOT solve the potential fragmentation issues (it has no way to defragment), but it does seem like a good implementation of malloc/free that attempts to reduce fragmentation as much as possible.

pfalcon, 2015-05-03:
> Well that's what I want to discuss! I have a working early implementation!

Nice, if you have it, please submit it as a pull request like everyone else does, and we'll discuss it.

As for hangouts, I have enough them at the day job. I'm ok with simpler schedule for hobby projects.

pfalcon, 2015-05-03:
> I think it would be a good place to start.

Good place to start what? As we discussed above, the no.1 problem uPy has is that it can't use malloc/free for memory management. Otherwise, there's malloc/free implementation in any ANSI C compiler (because it's part of standard). One implementation may be better than another, but until any of them can be used at all, there's not much difference.

cloudformdesign, 2015-05-04:
@pfalcon I meant a good place to start to implement **tinymem** -- which won't work with standard ANSI mallocs because it requires pointer indirection. And I can't just "submit a pull request" because it is two separate pull requests. Therefore I'm trying to get this pull request together first to implement malloc/free with **umm_malloc**.

I'm looking at the garbage collection code now. I think it should be doable for me to use umm_alloc to get a standard malloc/free gc working.

Also, I want to make a note. When I said that "it is going to have significant memory ramifications right off the bat" I was talking about this little gem: 

```C
    for (mp_uint_t block = 0; block < MP_STATE_MEM(gc_alloc_table_byte_len) * BLOCKS_PER_ATB; block++) {
        switch (ATB_GET_KIND(block)) {

```

This happens at several points in the code (line 231 is one place), and it basically assumes that all memory is allocated by gc.c, and that all memory falls on the BYTES_PER_BLOCK boundaries. It also assumes that it has direct access to the memory pool of the memory allocator, because it is stepping through it.

These things save memory, but they are not portable. However, these assumptions could easily work with both **tinymem** and **umm_alloc** -- are the assumptions worth it moving forward?

cloudformdesign, 2015-05-05:
I just realized that this doesn't apply to the "standard" way to document. Does upython have a "stanard" documentation? Have you thought about standardizing on doxygen?

I (try to) follow the contiki documenting style, which follows doxygen. Have you thought about using this? If there is no standard, I would suggest it as a good template:

https://github.com/contiki-os/contiki/blob/master/doc/code-style.c

cloudformdesign, 2015-05-07:
I'm thinking that all memory managers should go inside of the `lib/` folder, instead of `py/`. pfalcon has already suggested that I create a second memory manager which is built entirely off of std C malloc/free -- I don't think both of these files should be in the `py/` folder.

Any suggestions on the best way to do this / where the memory managers should go?

pfalcon, 2015-05-07:
> all memory managers should go inside of the lib/

No, lib/ is for general-purpose (enough) and reusable code. uPy memory managers are still uPy-specific, and py/ is the best place for them without inventing too much stuff at once.

cloudformdesign, 2015-05-07:
sounds good. Maybe if we get to many we can re-factor them into a "mem" directory or something later, if there becomes too many memory managers :P

dpgeorge, 2020-06-09:
Abstracting out the memory manager is a great goal.  But this issue is old and no progress has been made so I'll close it.

closed 2020-06-09 by dpgeorge