micropython/micropython #1170: New release - 1.3.11/1.4?

pfalcon, 2015-03-26:
1.3.10-282 tells that it would be nice to have a new release (maybe should have 2 already). And I guess it doesn't make sense to go 2-digit on micro version, and it can be 1.4 (90% coverage alone would warrant that). Previous TODOs for 1.4 can be moved to 1.5.

P.S. And would be nice to announce it on Python list(s).

dpgeorge, 2015-03-27:
Reason #1061 is held up is because new USB code is still untested on Windows and Mac, and HID stuff is reported to be buggy, eg #1123.  Further, the low power API is not yet agreed upon.

Anyway, that stuff is not going to come quickly, and in order not to hamper other development efforts I agree we should just push forward with a new release.

dpgeorge, 2015-03-29:
It's now v1.4.  Remains to announce on a Pthon list.

pfalcon, 2015-04-05:
Thanks @dpgeorge for sending the announcement!

closed 2015-04-05 by pfalcon